microsoft/AirSim #2977: Incorrect FOV in simGetCameraInfo call

clintlombard, 2020-08-26:
Using the Python API, changing the `FOV_Degrees` field in `settings.json` or using `simSetCameraFov()` to set a cameras FOV works. However, when querying the camera information with `simGetCameraInfo()`, it always returns a `CameraInfo.fov = 90`. Could this be incorrectly reporting the vertical FOV, or is it not being updated with the new horizontal FOV changes?

Python: 3.6.9
AirSim Plugin: commit bd8ee6be4183c28d402e4c34cce114312da46358 (latest on master at the time)
AirSim Python API: 1.2.8
UE: 4.25.3 (also tested in 4.24.3)
OS: Ubuntu 18.04 (docker container)

VC86, 2020-08-26:
I confirm the issue on 
Python 3.6
AirSim built on Windows from latest master commit: bd8ee6be4183c28d402e4c34cce114312da46358
UE: 4.24.3
OS: Windows 10

MoBaT, 2021-01-05:
Also seeing this issue on:

AirSim Plugin: commit f36efa6f3ed0b496cc8f551f2b20349d164f51bb
UE: 4.25.3
OS: Ubuntu 18.04

rajat2004, 2021-01-07:
Will be great if someone can test #3278 and see if it fixes the problem

desireB, 2021-04-15:
Hi Clint,

Would you mind sharing your settings.json ?

Thank you.

rajat2004, 2021-04-15:
@desireB #3573 might be useful for some examples of settings

desireB, 2021-04-15:
Thanks, @rajat2004 !

closed 2021-05-20 by zimmy87

Shromm-Gaind, 2022-12-12:
I posted this issue on unrealcv, however, I am not sure if it is maintained or checked. Can someone please explain to me which format of the projection matrix does AirSim generate? Currently, I am getting;  
[   0.0, 1.0,  0.0,   0.0],
[   0.0, 0.0, -1.8,   0.0],
[   0.0, 0.0,  0.0, 10.0],
[  -1.0, 0.0,  0.0,   0.0]
But I am not sure about this...It seems wrong. I was expecting to get;
[image]

PR #3278 (2021-05-20): Update Camera FoV as well when using simSetCameraFoV API 

Fixes #2977

## About

- Fixes camera FoV not being updated when using the API
- Adds example script for usage of API

Currently, only Render texture FoVs are being updated. This results in an incorrect FoV in the Camera info. **Note** that this is a behavioural change, previously when changing the FoV, the main viewport did not change, only the rendered images were affected. This change will cause the viewport to also change if the camera is the main one, such as the front center camera 

## How Has This Been Tested?

Using the script [... cut]