microsoft/AirSim #2979: Recording external camera frames

DanielGilo, 2020-08-26:
Hi, I know this was discussed recently on #2580 but the suggested solution  - pixel streaming - is not what I'm looking for. I don't need to stream anything to any device or network or browser. I simply want to capture frames of what I'm seeing at the editor from the external camera - the drone flying from third person perspective. Just like capturing scene images from regular front facing camera. Is there any way to do this via AirSim API or in any other way?

ivannson, 2020-09-18:
You can add a new camera and change the camera location settings in `settings.json` to be behind and above the drone. Experiment with x and z values and see where it gets you. Set something like x = -10, z = -5 first and see how the view turns out.

jonyMarino, 2020-11-06:
Hi @DanielGilo! Did you try ivannson approach?

jonyMarino, 2020-11-30:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-30 by jonyMarino