microsoft/AirSim #2980: Problem using depth images to create point cloud

GimpelZhang, 2020-08-27:
Hi all, 

As mentioned in the document of AirSim [Image APIs - AirSim](https://microsoft.github.io/AirSim/image_apis/#available_cameras): 

`For example, you may be able to feed perspective depth to ROS package such as depth_image_proc to generate a point cloud`

we can use depth images from AirSim to generate XYZ point cloud, or even XYZRGB point cloud. 

However, when I fed depth image & related camera info rostopics to the depth_image_proc/point_cloud_xyz node, I got the following WARNs, and no point cloud came out:

```
[ WARN] [1598496524.956492599]: [image_transport] Topics '/depth_image' and '/depth_camera_info' do not appear to be synchronized. In the last 10s:
	Image messages received:      25
	CameraInfo messages received: 0
	Synchronized pairs:           0
```

* OS: ubuntu 16.04
* unreal 4.24.3
* AirSim: master at [bd8ee6b](https://github.com/microsoft/AirSim/tree/bd8ee6be4183c28d402e4c34cce114312da46358)
* ros: kinetic

GimpelZhang, 2020-08-27:
I found in the ROS wrapper, camera_info topics are getting the current ros time as header stamp, while image topics are using the time from ImageResponse. 

After I made them the same, point cloud occurred as expected.

TooCoo, 2021-04-21:
I've been having a similar issue, I managed to fix the timestamps. But then the projection of the pointcloud is off, feels scaled (too small) and warped (not flat).

PR #2983 (2021-07-21): [ros] camera_info header stamp modified
Fixes: #2980

The header stamps of camera_info rostopics are modified. So timestamps of image topics and related camera_info topics are the same. 

Issue described here: #2980
closed 2021-07-21 by jonyMarino