microsoft/AirSim #2982: Problem building AirSim on Arch with anaconda3

iwishiwasaneagle, 2020-08-27:
Very similar to #2708, however the steps outlined there still result in errors. For example `export CXX="clang++-8" only works if you also run `sudo ln -sf /opt/anaconda3/envs/airsim/bin/clang-8 /opt/anaconda3/envs/airsim/bin/clang++-8`. However, the `./build.sh` runs smoothly until the following error is raised:
```
[ 83%] Linking CXX executable ../../output/bin/MavLinkTest
/usr/bin/ld: CMakeFiles/MavLinkTest.dir/opt/AirSim/MavLinkCom/MavLinkTest/Commands.cpp.o: undefined reference to symbol '__cxa_pure_virtual'
/usr/bin/ld: /opt/anaconda3/envs/airsim/bin/../lib/./libc++abi.so.1: error adding symbols: DSO missing from command line
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [MavLinkCom/MavLinkTest/CMakeFiles/MavLinkTest.dir/build.make:134: output/bin/MavLinkTest] Error 1
make[1]: *** [CMakeFiles/Makefile2:380: MavLinkCom/MavLinkTest/CMakeFiles/MavLinkTest.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Linking CXX static library ../output/lib/libAirLib.a
[ 85%] Built target AirLib
make: *** [Makefile:149: all] Error 2
```

## System
 - `uname -a` -> `Linux jhdesktop 5.7.15-1-MANJARO #1 SMP PREEMPT Tue Aug 11 15:00:37 UTC 2020 x86_64 GNU/Linux`
 - Fresh clone in `master` branch of microsoft/AirSim

## Steps to reproduce
 - Clone AirSim
 - `conda create --name airsim`
 - `conda activate`
 - `conda install -c conda-forge boost clang=8.0.1 libcxx=8.0.1`
 - `sudo ln -sf /opt/anaconda3/envs/airsim/bin/clang-8 /opt/anaconda3/envs/airsim/bin/clang++-8 # I think this step is the issue but install clangxx from conda-forge just creates a symlink to clang-8`
 - `./build.sh`

iwishiwasaneagle, 2020-08-27:
Relevant files:

[`./build.sh 2>&1 | tee build.log`](https://github.com/microsoft/AirSim/files/5136833/build.log)

[build.sh](https://github.com/microsoft/AirSim/files/5136835/build.txt)

[setup.sh](https://github.com/microsoft/AirSim/files/5136836/setup.txt)

rajat2004, 2020-08-27:
Strange, linker errors are always painful
Maybe try downloading `clang-8` binaries from https://releases.llvm.org/download.html and using those?

iwishiwasaneagle, 2020-08-27:
> Strange, linker errors are always painful
> Maybe try downloading `clang-8` binaries from https://releases.llvm.org/download.html and using those
> 
> Edit: Didn't read the description properly, the symlink of `clang` to `clang++` seems the most likely cause, since it's then compiling C++ with a C compiler

Hi there, it surely can't be the fact that I'm using clang and clang++ because the binaries from your link (for clang and clang++) are identical. See the screenshot below:
[image]

rajat2004, 2020-08-27:
Yup, you're correct, `clang++` is actually a symlink to `clang`, and the difference lies in the fact that invoking it by `clang++` tells it to link against C++ libraries. Just checked on my system as well.
Still, maybe try using the downloaded binaries and see if it works

iwishiwasaneagle, 2020-08-27:
> Yup, you're correct, `clang++` is actually a symlink to `clang`, and the difference lies in the fact that invoking it by `clang++` tells it to link against C++ libraries. Just checked on my system as well.
> Still, maybe try using the downloaded binaries and see if it works

So I tried the binaries, and a million other methods. In the end using g++ instead of clang++ has made it work for now...

rajat2004, 2020-08-27:
That's definitely strange, building with GCC was dropped since it was broken and didn't work, the Blocks env is up and running?

iwishiwasaneagle, 2020-08-27:
> That's definitely strange, building with GCC was dropped since it was broken and didn't work, the Blocks env is up and running?

Sadly not... Blocks and AirSim are supposedly not compatible with UE4.

rajat2004, 2020-08-29:
Just noticed that the issue was reopened
Are you trying to use OpenGL with the source-built UE? In that case, it needs to be uncommented in some UE file, and then engine needs to be rebuilt
Try running the binaries also to make sure there are no driver problems

If the clang problem still isn't fixed, you could try building clang, libcxx from source and use that. AirSim earlier used that, clang-5 was built from source and used. The PR which removed that was #2500, that would have llvm-5, commands can be easily updated to 8

Hope this helps!

iwishiwasaneagle, 2021-07-19:
Getting this off my `issuse` log

closed 2021-07-19 by iwishiwasaneagle