microsoft/AirSim #2987: Neighborhood Sim Minimizes Off-Focus

DamianRivas, 2020-08-28:
OS: Ubuntu 20.04
Release: Neighborhood v1.3.1

To reproduce the issue:
1. Run the neighborhoods.sh sim
2. Click on another window to go out of focus
3. The simulation "minimizes" or hides from view.

I am running the Neighborhood Simulation on a multi-monitor setup and I can't work on anything and view the simulation at the same time because the window minimizes every time it goes off focus. Is there a way to prevent this behavior?

There doesn't seem to be any other open nor closed issues on this.

rajat2004, 2020-08-29:
Not exactly sure what's happening, maybe a quirk of multi-monitor? If you are running the binary with the `-windowed` argument, maybe try without it.
Is something like below what you're looking for?
[image]

DamianRivas, 2020-09-09:
Hmm actually I'm not using the `-windowed` argument and perhaps that's what I need. I'll give that a try and see how it goes.

DamianRivas, 2020-09-25:
Forgot to report back but running the sim in windowed mode prevents the behavior. Good enough for me!

closed 2020-09-25 by DamianRivas