microsoft/AirSim #2988: using simEnableWeather(True) causes crash

ibrahim0v, 2020-08-28:
I am a customly packaged AirSim binary for Blocks environment. 

My versions are as follows:
AirSim : 1.3.1 Linux release
UE4    : 4.22
OS      : Ubuntu 18.04 LTS

Previously it would not open weather config menu when I press F10. So today I tried to use enable weather effects using the Weather Python API. But as soon as I reach the line where I implement simEnableWeather(True), AirSim crashes with the following messages:

```
24.0-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Unable to read VR Path Registry from /home/gygl/.config/openvr/openvrpaths.vrpath
Unable to read VR Path Registry from /home/gygl/.config/openvr/openvrpaths.vrpath
Unable to read VR Path Registry from /home/gygl/.config/openvr/openvrpaths.vrpath
Disconnecting mavlink vehicle
Waiting for mavlink vehicle...
Disconnecting mavlink vehicle
Waiting for TCP connection on port 4560, local IP 127.0.0.1
PX4
LowLevelFatalError [File:Unknown] [Line: 988] 
Node /AirSim/Weather/WeatherFX/WeatherGlobalParams 0 Package_LoadSummary    has zero prerequisites, but has not been queued.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=127232 LargeMemoryPoolOffset=258368 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
Have anyone faced this problem? And is there anyone that could help?

rajat2004, 2020-08-29:
So I did have a go at this, it's turning out to be a very tricky bug

First, the crash which happens when [`weather.py`](https://github.com/microsoft/AirSim/blob/master/PythonClient/computer_vision/weather.py) is used with the current latest binary- 
```
Assertion failed: IsInGameThread() [File:/home/ue4/UnrealEngine-4.24.2-linux-all-glsl/Engine/Source/Runtime/CoreUObject/Private/UObject/UObjectGlobals.cpp] [Line: 1065] 
Unable to load /AirSim/Weather/WeatherFX/WeatherGlobalParams. Objects and Packages can only be loaded from the game thread.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
```

Okay, quite straightforward error message, need to run in GameThread. Interesting note, this doesn't happen in the Editor, I even tried calling `simEnableWeather(true)` hundreds of times, didn't crash for me atleast.
So changed the code to run in game thread, all good. Now things got messy very quickly

On Packaging (this is another huge, time-consuming rabbit hole) came the biggest problem, the Weather assets don't get packaged at all! The Packaging tool determines which assets are to be cooked and packaged, and the Weather ones are loaded using the C++ code rather than blueprint using `StaticLoadObject()`, which doesn't reference the file for cooking.

Following some UE4 forum posts, Stackoverflow, tried out different ways, saved the assets again, used `FObjectFinder` to load the assets, etc but didn't work out. Right now the only option I can see left is to try to load it in the blueprint itself. I have 0 experience in working with blueprints though, will be interesting.

Just posting here for others to follow and to get some ideas to how to proceed. Let me know if anyone would like to know more about what did and didn't work, have a look at the code, etc

ibrahim0v, 2020-09-02:
@rajat2004 thnx for pointing me out to the right direction mate :) I have solved this issue.

I have realized that unreal engine, due to some reason (maybe because it is a Cpp Add-on on AirSim pluging),cannot reference WeatherFX throughout the project. Therefore it does not cook it or add it in the packaging of the game. However, unreal engine allows adding some unreferenced directories to cooking targets using -CookDir argument (or "Directories to Always Cook" in the GUI/Editor settings). This kind of forces cooking and packaging of the indicated directory/-ies. This way I have managed to cook the weather effects along with the whole project. 

`
RunUAT.sh BuildCookRun -platform=Linux -noP4 -cook -allmaps -build -stage -deploy -prereqs -CookDir="<airsim_src_dir>/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather" -archive -archivedirectory=<pkging-dir>/Blocks/ -project=<airsim_src_dir>/Unreal/Environments/Blocks/Blocks.uproject  
`

rajat2004, 2020-09-02:
Oh, didn't know about this that's one way of fixing the problem, thanks for the info!
It's still kind of a hack since it requires explicitly adding the directory, but works so that's nice. Would be better to fix this in AirSim itself

ibrahim0v, 2020-09-02:
Yeah you are right. A fix through AirSim would be great! 

I am closing this issue as it has been met.

zimmy87, 2021-04-08:
Given that the cause of this issue wasn't addressed, we are re-opening this to take a fix in the source

PR #3009 (2021-07-23): Fix Weather API crash
Fixes: #2988

With the latest release binary, using the `simEnableWeather` API causes crash, with the following error -
```
Assertion failed: IsInGameThread() [File:/home/ue4/UnrealEngine-4.24.2-linux-all-glsl/Engine/Source/Runtime/CoreUObject/Private/UObject/UObjectGlobals.cpp] [Line: 1065] 
Unable to load /AirSim/Weather/WeatherFX/WeatherGlobalParams. Objects and Packages can only be loaded from the game thread.
```

Note that it'll work fine in the Editor, with or without the fix and the next problem described below

Quite clear, and so added the `RunOnGameThread` part [... cut]
closed 2021-07-23 by zimmy87