microsoft/AirSim #2989: Are new features and refactoring being accepted?

r2d2Proton, 2020-08-28:
Forgive me, there appears to be some pull requests that have been sitting out there for a bit. I haven't looked at all the details but some pull requests looked trivial but not incorporated into the main branch (I believe).

Ideally, if I take the time to make and test changes I would like them to be considered for inclusion in a timely fashion.

Specific changes to the UnrealEngine side I recommend include removing the hardcoded Input mappings defined in ACarPawn::setupInputBindings and other such places.

Would this be of value to anyone using AirSim?

ash-xyz, 2020-08-29:
This! I want to spend some time updating the repo for ROS 2, building upon @rajat2004 Ubuntu 20.04 Noetic pull requests. However, my pull could very well be ignored if this is a dead repo and I'd hate to see this repo gone.

xxEoD2242, 2020-08-29:
I'm not one of the maintainers of this, but they just released a ton of updates in July and a major new release in April. Since Mircosoft is the one who pays the people to work on this, I imagine they are busy with other projects. But, this is actively updated on a near regular basis.

rajat2004, 2020-08-29:
I definitely wouldn't say that this is a dead repo, several features are being discussed and worked upon by users, and maintainers also triaging issues.
Just adding to @xxEoD2242's comment, it can take some time for the PRs being merged, I've generally seen that many things get merged quickly in a short span of time, about every 1-1/2 months maybe.  Can't say for sure when the PRs will get merged and all, that's up to the maintainers, since they are also probably very busy with other things, but definitely any contributions would be very much welcome :)

jonyMarino, 2021-01-07:
Hi @r2d2Proton! Of course, all contributions towards improving quality code are welcome. Sorry if there are times when we can not attend some pull request. It is not the general case. You can see all the PRs that we have merged since you opened this issue.

closed 2021-05-04 by jonyMarino