microsoft/AirSim #2990: One environment, multiple kinds of vehicles.

sandyUni, 2020-08-29:
I am reading the source code of  Airsim to add relative ranging sensor ([link](https://github.com/sandyUni/AirSim)).  After reading the official manual, I thought Airsim supports multiple vehicles well. However, as I dive into the source code and the mechanism of supporting multiple vehicles of Airsim, I found it does not have a game mode that can make the car and quadrotor coexisted. 

Since UE4 has a built-in physic model for groud car, Airsim make different game mode for  car and quadrotor in `ASimHUD::createSimMode`: (in [repo](https://github.com/microsoft/AirSim/blob/bd8ee6be4183c28d402e4c34cce114312da46358/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L294))

```cpp
void ASimHUD::createSimMode()
{
    std::string simmode_name = AirSimSettings::singleton().simmode_name;

    FActorSpawnParameters simmode_spawn_params;
    simmode_spawn_params.SpawnCollisionHandlingOverride = ESpawnActorCollisionHandlingMethod::AdjustIfPossibleButAlwaysSpawn;

    //spawn at origin. We will use this to do global NED transforms, for ex, non-vehicle objects in environment
    if (simmode_name == "Multirotor")
        simmode_ = this->GetWorld()->SpawnActor<ASimModeWorldMultiRotor>(FVector::ZeroVector, 
            FRotator::ZeroRotator, simmode_spawn_params);
    else if (simmode_name == "Car")
        simmode_ = this->GetWorld()->SpawnActor<ASimModeCar>(FVector::ZeroVector,
            FRotator::ZeroRotator, simmode_spawn_params);
    else if (simmode_name == "ComputerVision")
        simmode_ = this->GetWorld()->SpawnActor<ASimModeComputerVision>(FVector::ZeroVector,
            FRotator::ZeroRotator, simmode_spawn_params);
    else {
        UAirBlueprintLib::ShowMessage(EAppMsgType::Ok, std::string("SimMode is not valid: ") + simmode_name, "Error");
        UAirBlueprintLib::LogMessageString("SimMode is not valid: ", simmode_name, LogDebugLevel::Failure);
    }
}
```
This above code only runs one time at each simulation, therefore car and quadrotor or any other kinds of game mode can not coexist. This issue greatly limits its application scenarios such as groud-air cooperative tracking. 

Is there any plan for supporting different kinds of vehicles?

jonyMarino, 2020-09-02:
duplicate of #506

sandyUni, 2020-09-12:
yes, it is a  duplication  of #506

closed 2020-09-12 by sandyUni