microsoft/AirSim #2991: Tutorials and Walkthrough

JacopoPan, 2020-08-30:
I tried tor write a couple _extremely_ simple, self-contained tutorials on

- [Using AirSim's APIs / Compiling UE4 and AirSim](https://github.com/JacopoPan/a-minimalist-guide/blob/master/Part3-Using-AirSim.md)
- [Modifying/recompiling AirSim's Unreal plugin and using custom UE4 projects and meshes](https://github.com/JacopoPan/a-minimalist-guide/blob/master/Part4-Modifying-AirSim.md)

If anyone wants to check them out/correct me on anything I did wrong 

I was motivated in part by the fact that I found the documentation on https://microsoft.github.io/AirSim/ to be a bit scattered but, more importantly, some of it is as old as 5 years, while some was updated as recently last month: **is there any plan to re-factor/simplify/make current the existing body of documentation?**

jonyMarino, 2020-08-31:
@JacopoPan excellent initiative! We are very interested in these types of contributions. I will be looking at it and coming back to you soon.

JacopoPan, 2020-09-08:
@jonyMarino I was happy to see this had positive feedback: if you think there are certain parts I should/could polish and make into a PR, please let me know

jonyMarino, 2020-09-15:
@JacopoPan It seems to us that this documentation corresponds to different sections of the current documentation. We will see it well and return to you

JacopoPan, 2020-09-15:
Yes, of course: there are some additional terminal snippets, information I learned from open and closed Issues, and the custom pawn instructions and RLlib integration are new) but, as I wrote in the first message, one of the (I believe) limits of the current documentation is how scattered across pages it currently is (and that's what I think it would be useful to address)

jonyMarino, 2020-10-01:
@JacopoPan Sorry for the delays in incorporating your contribution. We are making efforts so that the contributions of the community are incorporated more quickly. Could you make a pull request for this and we continue there? It will get more attention if you do that