microsoft/AirSim #2992: "api call was not received entering hover mode for safety"

nmuchaxo, 2020-08-31:
Hello, I'm new to AirSim and I'm trying to use it with the ROS Wrapper. 
I did a script where I subscribe to /cmd_vel and publish the info on /airsim_node/drone_1/vel_cmd_body_frame.

Now, here is the isssue: When I start AirSim, I click "Play" and I run a script for the drone to take off. It takes off ("request ApiControl was successful" and "Vehicle is already armed" messages pop up) and then I get the following message  "api call was not received entering hover mode for safety" and everytime I try to send info through /cmd_vel to publish on the airsim topic the drone goes in the X, Y, Z axis, wherever I tell it to, but goes back to the same position where it was. Imagine I send "(...)linear: x: 1 (...)" he goes from X=0 to X=1 and then back to X=0.

Does anyone have any idea why this is happening or how can I have that message not popping up/ having the Api always enable? TIA

closed 2020-10-27 by nmuchaxo

AlexanderZ3, 2021-03-04:
The same question, does anyone has suggestions? I met this problem in windows normal setting. My setting is UE2.5, Airsim 1.4.0, Win10.  I can run multirotor controlled by the Remote controller successfully, but when I call the API, through calling the project DreoneShell in the root Airsim.sln,  I met the same problem: It takes off ("request ApiControl was successful" and "Vehicle is already armed" messages pop up) and then I get the following message "api call was not received entering hover mode for safety" . When I input forward command along x,y,z axis, it does not move at all.

Maybe someone can give some anlysis, thanks a lot!

AlexanderZ3, 2021-03-04:
@nmuchaxo hello, do you solve this problem?

nmuchaxo, 2021-03-04:
Hello I was on Linux and I don't remember very well, because it was some time ago... But I think (for me) it had to do with this:
Whenever I published on /cmd_vel, I published only one time and right after the drone received this input, it would stop and those messages popped up because it was not receiving more information on the /cmd_vel and for "safety reasons" it would stop. Imagine in real life, if the drone is not receiving any more input, it stops, or hovers in the same position... Makes sense to me.
So, whenever I published something I started doing with a rate. For example "publish -r 10...", with this the drone was getting or receiving constant information and did not stop receiving it. For me, this solved it.

Maybe it is happening the "same" to you. The drone is not always receiving information and for "safety reasons" (altough it is a simulator) it stops and hovers?...

BF111111, 2021-12-30:
> The same question, does anyone has suggestions? I met this problem in windows normal setting. My setting is UE2.5, Airsim 1.4.0, Win10. I can run multirotor controlled by the Remote controller [...]

Have you solved the problem yet?

BF111111, 2021-12-30:
> The same question, does anyone has suggestions? I met this problem in windows normal setting. My setting is UE2.5, Airsim 1.4.0, Win10. I can run multirotor controlled by the Remote controller [...]

Have you solved the problem yet? I have the same problem.

Shivam-Kataria, 2023-01-25:
> Have you solved the problem yet? I have the same problem.

I have the same issue. Has you been able to solve it?