microsoft/AirSim #3001: Bug: Blocks could not be complied on Ubuntu 18.04 with Unreal Engine 4.24.3

machinil, 2020-09-04:
/home/machini/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/WorldSimApi.cpp:116:40: error: no member named 'NameMode' in 'FActorSpawnParameters'
                new_actor_spawn_params.NameMode = FActorSpawnParameters::ESpawnActorNameMode::Required_ReturnNull;
                ~~~~~~~~~~~~~~~~~~~~~~ ^
/home/machini/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/WorldSimApi.cpp:116:74: error: no member named 'ESpawnActorNameMode' in 'FActorSpawnParameters'
                new_actor_spawn_params.NameMode = FActorSpawnParameters::ESpawnActorNameMode::Required_ReturnNull;
                                                  ~~~~~~~~~~~~~~~~~~~~~~~^
2 errors generated.
LogInit: Warning: Still incompatible or missing module: Blocks
LogInit: Warning: Still incompatible or missing module: AirSim

GimpelZhang, 2020-09-04:
See this: #3000 

NameMode is a member from higher version of UE.

saihv, 2020-09-04:
Merged https://github.com/microsoft/AirSim/pull/3000. @machinil Please pull master and try again.

jonyMarino, 2020-10-19:
@machinil I think this is not an issue any more

closed 2020-10-27 by machinil