microsoft/AirSim #3005: How to change settings.json? It become default setting when I turn on.

shkim1030, 2020-09-06:
Hi. I'm using windows, python version 3.8, unreal version 4.24.1

I'm doing some work at making auto-driving safety algorithm and, I'm learning airsim.

By the way, I want to change camera location and direction. 
So I tried changing settings.json, but it returns to default setting when I turn on airsim in unreal engine.
I even tried saving it with encoded with ASCII Code, but it doesn't work... How can I change my settings.json?

GimpelZhang, 2020-09-07:
The default path to the setting file is: /Documents/AirSim/settings.json

So are you changing this setting file? If so, please ignore my comment.

nabeelsherazi, 2020-09-22:
Either change the file in the Documents/Airsim folder, or, if you are trying to create a custom settings.json in the executable folder, make sure that it is in the same folder as the actual executable. This is _not_ the same as the top level executable in the binary folder. E.g. if you are using the Blocks environment, the settings.json file should be placed in `...\Blocks\Blocks\Binaries\Win64`

shkim1030, 2020-09-25:
> Either change the file in the Documents/Airsim folder, or, if you are trying to create a custom settings.json in the executable folder, make sure that it is in the same folder as the actual [...]

Thanks for your comment!
I made custom map named car, and project is located at D:\Unreal\car
I understood your comment as "Change Settings.json file which is in your project's Binaries -> Win64 folder."
However, in D:\Unreal\car\Binaries\Win64, there is no file named settings.json. 
So I copied 'settings.json' file in Document\Airsim folder, and pasted in D:\Unreal\car\Binaries\Win64, but it doesn't worked...

nabeelsherazi, 2020-09-25:
@shkim1030 When you start Airsim, it should list the location that it is pulling the settings.json file from in the top left corner logs. Can you see where that location is for your setup?

jonyMarino, 2020-11-06:
Hi @shkim1030! The place searched for the settings.json would be (in my windows machine) C:\Program Files\Epic Games\UE_4.24\Engine\Binaries\Win64\settings.json.
if that doesn't help, try debugging here please:
https://github.com/microsoft/AirSim/blob/d59ceb7f63878f5e087ea802d603ba0fd282ff56/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L365-L372

jonyMarino, 2021-01-12:
Hi @shkim1030! can we close this issue, or do you still have it?

closed 2021-02-01 by jonyMarino