microsoft/AirSim #3006: ComputerVision mode: How to set the Unreal Engine window image resolution to be the same as the camera's resolution ?

TomAvrech, 2020-09-06:
In the settings.json, I set the cameras to be 64x64.
In the Unreal engine window, not only does it have different resolution, but also a different aspect ratio (16:9 ?).
How can I make them be the same --> have the Unreal Engine's big window BE the ComputerVision mode's main camera ?
[image]
Settings.json:

{
	"SettingsVersion": 1.2,
	"CameraDefaults": {
		"CaptureSettings": [
		  {
			"ImageType": 0,
			"Width": 64,
			"Height": 64,
			"FOV_Degrees": 90,
			"AutoExposureSpeed": 100,
			"MotionBlurAmount": 0
		  },
		  {
			"ImageType": 1,
			"Width": 64,
			"Height": 64,
			"FOV_Degrees": 90,
			"AutoExposureSpeed": 100,
			"MotionBlurAmount": 0
		  },
		  {
			"ImageType": 2,
			"Width": 64,
			"Height": 64,
			"FOV_Degrees": 90,
			"AutoExposureSpeed": 100,
			"MotionBlurAmount": 0
		  },
		  {
			"ImageType": 3,
			"Width": 64,
			"Height": 64,
			"FOV_Degrees": 90,
			"AutoExposureSpeed": 100,
			"MotionBlurAmount": 0
		  }
		]
	},
	"SimMode": "ComputerVision"
}

clintlombard, 2020-09-09:
Have a look at the [Unreal Engine Command-Line Arguments Guide](https://docs.unrealengine.com/en-US/Programming/Basics/CommandLineArguments/index.html), in particular the `ResX` and `ResY` parameters.

ivannson, 2020-09-18:
I think you can also change it in editor settings manually

jonyMarino, 2020-11-06:
@TomAvrech, how it went?

jonyMarino, 2020-11-30:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-30 by jonyMarino