microsoft/AirSim #3010: ClockSpeed issues with PX4 SITL

clintlombard, 2020-09-08:
When changing the `ClockSpeed` parameter in `settings.json` to anything faster or slower than normal speed, flight becomes unstable with PX4 in SITL. I even tried setting PX4's `PX4_SIM_SPEED_FACTOR` to no avail.

This seems to be a long standing issue https://discuss.px4.io/t/px4-sitl-airsim-lockstep-and-timeouts/17380

Python: 3.6.9
AirSim Plugin: commit bd8ee6b 
AirSim Python API: 1.2.8
UE: 4.25.3
PX4: v1.10.1 and v1.10.2 (v1.11.0 doesn't yet work with AirSIm)
OS: Ubuntu 18.04 (docker container)

m1baldwin, 2020-09-14:
@clintlombard Just wondering, when you try running v1.11.0 of the px4 release, what errors do you get? I am getting accel timeout errors. When I try running with the v1.10.*, it works fine though.

clintlombard, 2020-09-14:
No errors on `v1.11.0` it just doesn't seem to connect. Might be the updates to mavlink api. In `v1.10.*` under the normal clock speed (1.0) I get the accel timeouts and things still work. However, if you decrease the clock speed to say 0.25, then the PX4 EKF GPS fusion just times out (so basically just dead reckoning).

m1baldwin, 2020-09-14:
Interesting, I can at least get it to connect with px4 version 1.11.0. 

When I have tried running at a speed factor of less than 1 with px4. I have also seen issues. For me, the issues I saw is that the drone was no longer able to hit waypoints accurately, i.e. it would usually completely overshoot them. I couldn't find a fix and haven't looked into why this might be happening.

Is decreasing the clock speed also decreasing the rate the sensor plugin in airsim for the gps is producing data? Could be that the px4 sitl stack is still running at full speed, however the airsim clock is reduced per the spec in the settings file. Maybe there is some signaling missing here to properly configure the px4 sitl to run at a matching clock rate?

Just a thought, there is an environmental variable that can be set to configure the px4 sim speed. Maybe this also needs to be set before running px4 sitl? It can be set with: 
export PX4_SIM_SPEED_FACTOR <R>
https://dev.px4.io/v1.10/en/simulation/gazebo.html#change-simulation-speed

clintlombard, 2020-09-14:
I'll have to give `v1.11.0` another try.

Well technically things should be running in lockstep so I would assume PX4 would just wait on AirSim.

I have tried the `PX4_SIM_SPEED_FACTOR` environment variable but this didn't have any noticeable affect.

clintlombard, 2020-09-15:
After trying `v1.11.0` again, I see that PX4 and AirSim do connect. However, it seem the params in `settings.json` aren't being sent to PX4 so it can't takeoff due to a missing control stick
```
INFO  [commander] Armed by external command
WARN  [commander] Failsafe enabled: No manual control stick input
INFO  [commander] Failsafe mode activated
INFO  [commander] Takeoff detected
WARN  [commander] Takeoff denied! Please disarm and retry
INFO  [commander] Failsafe mode deactivated
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing
```

m1baldwin, 2020-09-15:
@clintlombard One option here might be to connect to QGC as well when you launch the px4 sitl. You can edit the firmware params just like you would with an actual drone, and it should be preserved when you re-run px4_sitl. The only time these params are reset is when you run make clean before a fresh make px4_sitl command, or go to a different px4 sitl airframe.

Or, you can issue these commands: https://microsoft.github.io/AirSim/px4_sitl/#manually-set-parameters in the px4 sitl window and see if you can at least fly in offboard mode to check if that part of your stack is working. 

Probably, we need one of the devs for airsim to shed some light here on why the params might not be sent from airsim.

zimmy87, 2021-02-10:
Hi all, @kemen209's PR #3158 fixed this issue for me. Please retest with his PR applied and let me know if you are still seeing this issue.

closed 2021-02-10 by zimmy87