microsoft/AirSim #3014: Can I create a custom environment like City environment and control the other cars in the environment without having to create everything on unreal engine from scratch?

Josepharol, 2020-09-10:
@sytelus  @lovettchris 

What I am trying to do : Create a custom environment with different driving scenarios ( something similar to city environment) and have a couple of cars to follow a particular path and I will drive the airsim car in the environment.

What I have : I already have a blocks environment and can add one airsim vehicle.

Question 1 : If I add the settings for multiple vehicles in the Json file --> I have two airsim cars, and i can control both with the Logitech G920 steering I have. However, I want to control only one airsim car with steering and I want the other set of multiple cars in the environment on specific paths with particular speed. Does airsim support this function or should I manually create a spline follow path on unreal engine for all the other cars in the environment?

Question 2: Does airsim retrieve information only about the airsim car/drone that we drive in the environment? or is it possible to retrieve information about the other cars in the environment too? Do you have any documentation for adding additional cars/drone into the environment, control them, collect specific information about them?

Any help would be appreciated. I am right now creating other cars in the environment to follow a spline path on unreal engine with custom blueprints following - [https://www.youtube.com/watch?v=bWXI91FdMtk] and it is of course a tedious process to create everything from scratch. Just was wondering if there was any other easier way to achieve that?

sytelus, 2020-09-11:
AirSim settings allow the `RC` section that specifies which vehicle is controlled how:

```
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "RC": {
        "RemoteControlID": -1
      },
```
See more info here: https://microsoft.github.io/AirSim/settings/

You can control multiple vehicles by specifying vehicle name in API calls. More info here: https://microsoft.github.io/AirSim/multi_vehicle/

Josepharol, 2020-09-20:
@sytelus Thank you for your response. So kind of you to respond to my queries. And indeed your advice solve my problem. However, I still didn't explore the RemoteControlD, but I didn't mention the car I want to control with steering in the API coding, so I am controlling one car with the steering and other cars respond to API calls. Thank you. If in case, I need to have a different approach, please do let me know. Thanks

closed 2020-10-09 by jonyMarino