microsoft/AirSim #3015: How to skip the animation of Modular Neighborhood Pack

KailinTong, 2020-09-11:
Dear all,

I am new to the Airsim and Unreal Engine.

The Modular Neighborhood Pack works with Airsim. One small trouble is that Modular Neighborhood Pack has to play a one-minute animation at first and then starts to the Airsim game mode.

Do you have some tips on how to skip the animation?

Thanks a lot!

Best regards
Kailin

plushpluto, 2020-09-13:
Not sure if this will help. 
I guess it is a part of the cinematics that is usually placed for an introduction in a game or a scene. Anyways, you need to find a way to disable "matinee". I did encounter something like this but that was for UE4-21 (which is not available in UE4-25). Not sure what is your version.

Check out these:
https://docs.unrealengine.com/en-US/Engine/Matinee/UserGuide/index.html
https://answers.unrealengine.com/questions/84997/view.html
https://docs.unrealengine.com/en-US/Engine/Sequencer/HowTo/MatineeConversionTool/index.html

KailinTong, 2020-09-14:
Thank you, @plushpluto,

I am using Version: 4.24.3, and I could not find matinee in the editor.

Do you know a way to edit cinematics to disable the animation?

Best,
Kailin

closed 2020-09-14 by KailinTong