microsoft/AirSim #3019: Maximum nu of multi agents that can be supported by aisim

neeraj310, 2020-09-13:
Hi Support Team

I am working on a survey for Airsim capabilities and need few clarifications regarding the number of supported agents. 

a) What is the maximum number of agents that can be supported on a normal desktop(16GB RAM, GPU)

b) Is it possible to add custom vehicles. I want to design my own vehicle and use it inside Airsim framework. 

Neeraj

neeraj310, 2020-09-20:
Hi Team 

Can you please revert back on this issue.

Neeraj

xxEoD2242, 2020-09-20:
I think your first question depends on what you want to accomplish. I've worked with up to 20 drones so far and never had a problem. It depends a lot of what you are asking AirSim to provide you.

For example, taking pictures is inherently a costly operation and if you want to take pictures of all 20 drones at say 10 Hertz sampling rate, then you will probably slow down the simulation considerably. Other people may have different experiences but with 16 GHz of RAM, that's what I've seen. What kind of GPU can make a difference as well.

For you second question, it is possible. Take a look at #2955 and see the branch that the person has been working on to implement a fixed-wing aircraft.

closed 2021-02-01 by jonyMarino