microsoft/AirSim #3021: Speeding up AirSim

TomAvrech, 2020-09-14:
Using the ComputerVision mode, I plot a lot of lines, and therefore the simulator is very slow.

How can I run AirSim in a way that will maximize the speed of everything ?

Is AirSim already running on maximum speed as the default mode ?

agentfeux, 2020-09-14:
You can try to clock the speed up in the setting file like this exemple:
`	"SettingsVersion":1.2,
	"LogMessagesVisible":true,
	"ClockSpeed": 1,`

TomAvrech, 2020-09-15:
> 
> 
> You can try to clock the speed up in the setting file like this exemple:
> ` "SettingsVersion":1.2, "LogMessagesVisible":true, "ClockSpeed": 2,`

Thanks, but even with "ClockSpeed": 10, nothing seems to improve.

ivannson, 2020-09-18:
First of all, check that you don't have 'Use less CPU in the background' enabled in Unreal. 

I wouldn't think speeding up the clock speed would help if your computer is struggling to process everything as it is. It allows to run 1 hr of simulation in 30 min of real-time for example, but only if the hardware can handle it. 

Things you can try:
- Lower the quality in scalability settings
- Reduce the window size (i.e. 360p will be quicker than 1080p)
- If running in editor, try launching in a separate editor window or as a stand-alone game
- Try no rendering mode if you are recording images with an API

jonyMarino, 2020-11-06:
Hi @TomAvrech! Did you try ivannson suggestions?

closed 2021-03-22 by jonyMarino