microsoft/AirSim #3024: landscape mountain segmentation

keep9oing, 2020-09-17:
Hi, I'm studying using airsim and environments

I want to segment the road in the mountain. such as below

[image]

However, the segmentation is done like landscape terrain and road is the same object. How can I separate them in the scene?

ykhedar, 2020-09-18:
I think you are only able to see objects in the segmentation view which are static meshes, like the trees, bridges, stones etc.. the dirt road is actually part of the terrain(landscape) itself. So you would not see it separate from the landscape terrain.

kharyal, 2020-11-07:
While trying to run AirSim on environments from the Unreal Marketplace, I am facing the following issue in getting correct segmentation images
[image]
[image]

I want separate segmentation for road and believe that the above issue is happening because the meshes are stored in a Blueprint Class Actor. Is there any way to penetrate through the BP to get correct segmentation?

TIA

kcaisfather, 2021-02-01:
Is there any solution for this inseparable segmentation issue. I have the same problem. We want to train our model to find drivable area with semantic segmentation method. Is there any opinion to deal with this in UE platform?

kharyal, 2021-02-01:
I did that by porting the whole environment to Unity, separating individual meshes and then importing it back to Unreal engine. But, it has its own set of problems like textures might not appear in Unity.

-Regards

closed 2021-03-22 by jonyMarino