microsoft/AirSim #3025: Can I use API to get images every frame in the game?

c6376315qqso, 2020-09-17:
I fixed my game frame rate to 30 fps.
Now I want to get images (depth map or segmentation or whatever) using python API in every frame, Is this achievable?

GimpelZhang, 2020-09-18:
From my point of view, 30 Hz of images capture is quite a high demand on your hardware.

c6376315qqso, 2020-09-18:
> From my point of view, 30 Hz of images capture is quite a high demand on your hardware.

Yeah,it is. But I want to know how to achieve getting images every frame?

ivannson, 2020-09-18:
I think the speed with which you are getting images depends on your hardware, you can time the process to determine the camera frame rate. If you strictly need it every frame, rather than just as quickly as possible, then you can set fixed frame rate within the Editor settings. 

What I mean is this:
Suppose it takes 0.1 seconds to get an image using the Python API, so your recording frame rate is 10 fps, whereas your simulation is running at 30 fps. I'm not sure you can speed the process up and obtain images quicker than 10 fps as it's limited by the hardware, but you can decrease the editor frame rate to 10 fps, so you are getting images every frame.

That said, I'm not sure why exactly you'd want to do that. As opposed to Lidar, I wouldn't think it really matters if the images are collected every other frame.

c6376315qqso, 2020-09-18:
> I think the speed with which you are getting images depends on your hardware, you can time the process to determine the camera frame rate. If you strictly need it every frame, rather than just as [...]

Emmm,I need to synchronize sampling in airsim with the tick function in ue4, Because I

c6376315qqso, 2020-09-18:
> I think the speed with which you are getting images depends on your hardware, you can time the process to determine the camera frame rate. If you strictly need it every frame, rather than just as [...]

Because I also get positions in tick function in ue4, so I want to exactly get images every tick, is this achievable?

niekvleeuwen, 2020-10-29:
I also looked at the possibility of receiving images every frame. I added an extra feature to AirSim for this purpose, see the pull request [here](https://github.com/microsoft/AirSim/pull/3102). Maybe you can take a look.

closed 2022-02-25 by jonyMarino