microsoft/AirSim #3026: Cannot figure out the right subwindow setting(settings.json) for multiple drones equipped with camera

skykim0609, 2020-09-17:
Hello. I am currently using AirSim 1.3 with Unreal 4.24 in Ubuntu 18.04 LTS version. I am trying to control 3 quadrotors in AirSim, and view the camera outputs through the subwindow. Following is the current settings.json file.

## 
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
      "Drone_1": {
          "VehicleType": "SimpleFlight",    
          "DefaultVehicleState":"Armed",        
          "X": 10, "Y":10.0, "Z":-20,
          "Yaw":0,
    "Cameras": {
              "camera1": {
                "CaptureSettings": [
                  {
                    "PublishToRos": 1,
                    "ImageType": 0,
                    "Width": 640,
                    "Height": 480,
                    "FOV_Degrees": 86,
                    "TargetGamma": 1.5
                  }
                ],
                "X": 0.50, "Y": 0, "Z": -0.10, 
                "Pitch": 0, "Roll": 0, "Yaw": 180
              }
          }
        },
"Drone_2": {
          "VehicleType": "SimpleFlight",  
          "DefaultVehicleState":"Armed",          
          "X": 20, "Y":10.0, "Z":-20,
          "Yaw":0,
    "Cameras": {
              "camera2": {
                "CaptureSettings": [
                  {
                    "PublishToRos": 1,
                    "ImageType": 0,
                    "Width": 640,
                    "Height": 480,
                    "FOV_Degrees": 86,
                    "TargetGamma": 1.5
                  }
                ],
                "X": 0.50, "Y": 0, "Z": -0.10, 
                "Pitch": 0, "Roll": 0, "Yaw": 180
              }
          }
        },
"Drone_3": {
          "VehicleType": "SimpleFlight",      
          "DefaultVehicleState":"Armed",    
          "X": 35, "Y":10.0, "Z":-20,
          "Yaw":0,
    "Cameras": {
              "camera3": {
                "CaptureSettings": [
                  {
                    "PublishToRos": 1,
                    "ImageType": 0,
                    "Width": 640,
                    "Height": 480,
                    "FOV_Degrees": 86,
                    "TargetGamma": 1.5
                  }
                ],
                "X": 0.50, "Y": 0, "Z": -0.10, 
                "Pitch": 0, "Roll": 0, "Yaw": 180
            }
      }
    }
},
"SubWindows": [
      {"WindowID": 0, "ImageType": 0, "CameraName": "camera1", "VehicleName": "Drone_1", "Visible": true},
      {"WindowID": 1, "ImageType": 0, "CameraName": "camera2", "VehicleName": "Drone_2", "Visible": true},
      {"WindowID": 2, "ImageType": 0, "CameraName": "camera3", "VehicleName": "Drone_3", "Visible": true}
    ]
}

##

I played this setting in the default given "blocks" environment. Following screenshot is the result.
[image]

The later 2 images displayed on the subwindows 1, 2 are apparently not the image from cameras on drone_2, drone_3.
I get "CameraID in <SubWindows> element in settings.json is invalid" error message, but I cannot figure out what is wrong with this setting.
Is there any way to fix this error?

rajat2004, 2020-09-18:
Probably should have added this in the docs, but the `VehicleName` element in the Subwindow settings is only there in master right now, and not in the latest release binaries

skykim0609, 2020-09-18:
> Probably should have added this in the docs, but the `VehicleName` element in the Subwindow settings is only there in master right now, and not in the latest release binaries

Thank you for reply. So for now, there isn't any way of displaying camera views from multiple drones in subwindows, right?

rajat2004, 2020-09-18:
Yup, not without building AirSim from source atleast. If needed, you could build binaries yourself after that

skykim0609, 2020-09-19:
I don't know what it takes to build my binaries so I'd rather consider other ways to deal with this problem. Thank you anyway.

closed 2020-09-19 by skykim0609