microsoft/AirSim #3027: Dynamically set sensors orientation/pose

yuyangch, 2020-09-17:
Hi! 
    Is there a way to dynamically change sensors (mainly Lidar)'s pose (mainly orientation), relative to the vehicle? 
    I did read the documentation listed:  https://microsoft.github.io/AirSim/lidar/
    But the to my understanding the lidar configuration through json is static. 

Thanks in advanced!

rajat2004, 2020-09-18:
Currently this isn't possible, but such an API can be added
It would need to modify the `relative_pose` in [`LidarParameters.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/lidar/LidarSimpleParams.hpp#L29)

Do you specifically need to change it relative to the vehicle, or just need to change the pose like move the lidar around? In the latter case, you could just use `simSetVehiclePose` and move the vehicle wherever required

yuyangch, 2020-09-18:
Thanks! I specifically need to change the relative pose, i want to keep the vehicle/drone operating normally.

I will look into implementing those changes, thanks for the pointers.

would you say the quick way to start is to look into the  "simPrintLogMessage API" commit? (attached below)
-----------------------------------------------------------------------------------------------------------------------------
I've added this API (albeit with name `simPrintLogMessage`).  You can see documentation here: https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#common-apis

This commit demonstrates the changes needed to add new APIs:
https://github.com/Microsoft/AirSim/commit/f0e83c29e7685e1021185e3c95bfdaffb6cb85dc

_Originally posted by @sytelus in https://github.com/microsoft/AirSim/issues/635#issuecomment-347402954_

rajat2004, 2020-09-18:
Yeah, that commit is a useful reference for making an API change, it's a bit old, but shows the main RPC related files where you would need to make changes. You can also have a look at https://github.com/microsoft/AirSim/pull/2810, that actually removes an API related to sensor data so can show more effectively which files you might need to modify.
There won't be any need to change the structs, etc. just need to add a method which changes the relative pose in the parameters and call that, several other methods in the RPC files which take `Pose` as input will be useful for reference. That PR also reduces some of the boiler-plate code to find the sensor, which possibly will be required.

yuyangch, 2020-09-18:
Thank you so much for the information, I really appreciate it!

eduardkieser, 2020-10-23:
Hi, @yuyangch Have you made any progress in this? I need the same for the rangefinder sensor. I.e. set sensor pose relative to the vehicle in real-time. Unfortunately I have almost no c++ experience, so any tips would go a long way.