microsoft/AirSim #3030: AirSim build error on Windows 10 -> AirLib.lib error LNK2001, LNK2019, LNK1120

rmolinabadia, 2020-09-20:
Hi everyone,

I'm trying to build AirSim on my computer:
- Windows 10.
- Visual Studio 2019, with Desktop Development with C++ and Windows 10 SDK 10.0.18362.
- AirSim v1.3.1.
- Unreal Engine 4.25.

I've followed the building instructions as stated in [https://microsoft.github.io/AirSim/build_windows/], but I do get an error after running the 'build.cmd' command. The issue comes from the AirLib.lib library, throwing LNK2001, LNK2019 and LNK1120 errors.

The attached image includes the entire output of the 'build.cmd'. Errors can be seen aprox. from the 3/4 onwards.

Steps to build AirSim are quite simple, but I don't get to understand why am I getting these errors every time I try to build it. Could anyone please help me?

Thanks in advance,
rmolinabadia

[image]

rajat2004, 2020-09-20:
Not sure what's happening, the command prompt also is correct otherwise that could have been causing the erorrs. Maybe run `clean.cmd` or `clean_rebuild.bat` once and try again
The CMake version seems a bit stragne, I'm using 3.16, the 3.14 one seems somewhat old, was that installed separately earlier? Even the latest 3.14 version is 3.14.7

rmolinabadia, 2020-09-20:
Hi @rajat2004 ,

As you said, CMake version update to 3.16.9 did the job.

Thanks for your fast reply and your help!

closed 2020-09-20 by rmolinabadia