microsoft/AirSim #3031: Distance sensor distance data is same for all cars? and it is oscillating?

Josepharol, 2020-09-20:
Problem:
I have 4 cars in my environment. And I am driving only car1 with steering input, other cars are stationery without any movement. But why is that they all give the same distance data as sensor output? And also, why is it oscillating?

As per my understanding, if the distance between two cars are calculated--> As I approach closer to another car, my distance value should reduce? But it doesn't in this model.

I am sure I am missing something and there are many chances that I might be wrong or my approach might be. Kindly requesting your guidance @sytelus 

Settings.JSON - find below

`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Vehicles": {
    "Car1": {
      "VehicleType": "PhysXCar",
      "AutoCreate": true,
      "X": -40,
      "Y": -31,
      "Z": 0,
      "Yaw": 180,
      "Sensors": {
        "Distance": {
          "SensorType": 5,
          "Enabled": true,
          "DrawDebugPoints": true
        }
      }
    },
    "Car2": {
      "VehicleType": "PhysXCar",
      "AutoCreate": true,
      "X": -46,
      "Y": -31,
      "Z": 0,
      "Yaw": 180,
      "Sensors": {
        "Distance": {
          "SensorType": 5,
          "Enabled": true,
          "DrawDebugPoints": true
        }
      }
    },
    "Car3": {
      "VehicleType": "PhysXCar",
      "AutoCreate": true,
      "X": -130,
      "Y": -31,
      "Z": 0,
      "Yaw": 180,
      "Sensors": {
        "Distance": {
          "SensorType": 5,
          "Enabled": true,
          "DrawDebugPoints": true
        }
      }
    },
    "Car4": {
      "VehicleType": "PhysXCar",
      "AutoCreate": true,
      "X": -136,
      "Y": -31,
      "Z": 0,
      "Yaw": 180
    }
  }
}`

Distance sensor API code:

`import airsim
import time

client = airsim.CarClient()
client.confirmConnection()

while True:
    data_car1 = client.getDistanceSensorData(vehicle_name="Car1")
    data_car2 = client.getDistanceSensorData(vehicle_name="Car2")
    data_car3 = client.getDistanceSensorData(vehicle_name="Car3")
    print(f"Distance sensor data: Car1: {data_car1.distance}, Car2: {data_car2.distance}, Car3: {data_car3.distance}")
    time.sleep(1.0)`

Output log:

> Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Distance sensor data: Car1: 0.11076667159795761, Car2: 0.11076667159795761, Car3: 0.11076667159795761
Distance sensor data: Car1: 0.006970573682337999, Car2: 0.006970573682337999, Car3: 0.006970573682337999
Distance sensor data: Car1: 0.22495973110198975, Car2: 0.22495973110198975, Car3: 0.22495973110198975
Distance sensor data: Car1: 0.23937900364398956, Car2: 0.23937900364398956, Car3: 0.23937900364398956
Distance sensor data: Car1: 0.15649870038032532, Car2: -0.17552848160266876, Car3: -0.17552848160266876
Distance sensor data: Car1: -0.190812349319458, Car2: -0.190812349319458, Car3: -0.190812349319458
Distance sensor data: Car1: 0.11031185835599899, Car2: 0.11031185835599899, Car3: 0.11031185835599899
Distance sensor data: Car1: 0.2529156804084778, Car2: 0.2529156804084778, Car3: 0.2529156804084778
Distance sensor data: Car1: 0.11173819750547409, Car2: 0.11173819750547409, Car3: 0.11173819750547409
Distance sensor data: Car1: 0.28879934549331665, Car2: 0.28879934549331665, Car3: 0.28879934549331665
Distance sensor data: Car1: -0.25463196635246277, Car2: -0.25463196635246277, Car3: -0.25463196635246277
Distance sensor data: Car1: 0.16131283342838287, Car2: 0.16131283342838287, Car3: 0.16131283342838287
Distance sensor data: Car1: 0.16709885001182556, Car2: 0.16709885001182556, Car3: -0.23638857901096344
Distance sensor data: Car1: 0.12369690835475922, Car2: 0.12369690835475922, Car3: 0.12369690835475922
Distance sensor data: Car1: -0.18763548135757446, Car2: -0.18763548135757446, Car3: -0.18763548135757446
Distance sensor data: Car1: 0.12698155641555786, Car2: 0.12698155641555786, Car3: 0.12698155641555786
Distance sensor data: Car1: 0.5347021818161011, Car2: 0.5347021818161011, Car3: 0.5347021818161011
Distance sensor data: Car1: -0.22548460960388184, Car2: -0.22548460960388184, Car3: -0.22548460960388184
Distance sensor data: Car1: 0.03354516625404358, Car2: 0.03354516625404358, Car3: 0.03354516625404358
Distance sensor data: Car1: -0.34821388125419617, Car2: -0.34821388125419617, Car3: -0.34821388125419617
Distance sensor data: Car1: 0.10065693408250809, Car2: 0.10065693408250809, Car3: 0.10065693408250809
Distance sensor data: Car1: 0.062413908541202545, Car2: 0.062413908541202545, Car3: 0.062413908541202545
Distance sensor data: Car1: -0.2628134787082672, Car2: -0.2628134787082672, Car3: -0.2628134787082672
Distance sensor data: Car1: -0.2646542191505432, Car2: -0.2646542191505432, Car3: -0.2646542191505432
Distance sensor data: Car1: -0.07307260483503342, Car2: -0.07307260483503342, Car3: -0.07307260483503342
Distance sensor data: Car1: -0.207633838057518, Car2: -0.207633838057518, Car3: -0.207633838057518
Distance sensor data: Car1: 0.2150786966085434, Car2: 0.2150786966085434, Car3: 0.2150786966085434
Distance sensor data: Car1: -0.10061611235141754, Car2: -0.10061611235141754, Car3: -0.10061611235141754
Distance sensor data: Car1: 2.657975435256958, Car2: -0.3295596241950989, Car3: -0.3295596241950989
Distance sensor data: Car1: 6.421899318695068, Car2: -0.13854628801345825, Car3: -0.13854628801345825
Distance sensor data: Car1: -0.32458552718162537, Car2: -0.32458552718162537, Car3: -0.32458552718162537
Distance sensor data: Car1: -0.20408090949058533, Car2: -0.20408090949058533, Car3: -0.20408090949058533
Distance sensor data: Car1: 0.1641518622636795, Car2: 0.1641518622636795, Car3: 0.1641518622636795
Distance sensor data: Car1: -0.277778297662735, Car2: -0.277778297662735, Car3: -0.277778297662735

rajat2004, 2020-09-20:
This was a problem with how the sensor was placed by default in Car vehicle, this bug along with several other additions to the distance sensir were done in #2807, which has been merged into master
You haven't specified which version of AirSim you're using, so I'm guessing the binary. Try out the master if you're able to and see if the problem's still occurring

Josepharol, 2020-09-20:
@rajat2004 Thank you for your response. I really appreciate it.  I built the airsim again according to [https://microsoft.github.io/AirSim/build_windows/](url) & also I am running it currently in the city environment V1.3.1 FROM [https://github.com/Microsoft/AirSim/releases](url) .  I feel like I have all latest releases. However when I run the `distance_sensor_multi.py`, i still get the same values for distance of all cars in the environment. In my settings.json file, the settings version, I tried `both "SettingsVersion": 1.2,` `"SettingsVersion": 1.3,` . I am not sure if that what you meant by latest release of airsim version?

rajat2004, 2020-09-20:
Yeah, so this isn't available in the 1.3.1 release, it's currently in master only, so you'll have to use the latest master built from source, and not the 1.3.1 binaries

Josepharol, 2020-09-20:
@rajat2004 I am sorry, this might be an amateur question. But, right now, I have built the latest airsim. I am going to use them in the City Environment.
When you say "use latest master built from source", should I download the any other versions of city environment and check it or can you please be specific on what I should do right now?

rajat2004, 2020-09-20:
Just follow the steps in https://microsoft.github.io/AirSim/build_windows/ and use the Blocks env
If you want to use the City env with latest master, you'll need to buy it from the Unreal Marketplace, since the code inside binaries can't be changed

Josepharol, 2020-09-20:
@rajat2004 Thank you. It works in the blocks environment. Do you know if any other sensors have this issue? That they work in blocks but not in the city environment? Just to know, if this is the case with all of the sensors?

rajat2004, 2020-09-21:
You can have a look at the commits in the master branch to see what changes have happened after the last release. Off the top of my head, there was a fix for Lidar `DrawDebugPoints` setting, and some camera fixes also

closed 2020-09-21 by jonyMarino