microsoft/AirSim #3032: python api calling camera capture in loop raise a breakpoint

Huibean, 2020-09-21:
Windows 10 platform
while get camera view in loop it raise a breakpoint in Visual Studio
```python
while True:
        responses = client.simGetImages([airsim.ImageRequest(0, airsim.ImageType.Scene, False, False)])
        response = responses[0]
        img1d = np.frombuffer(response.image_data_uint8, dtype=np.uint8) 
        frame = img1d.reshape(response.height, response.width, 3)
```

[image]

jonyMarino, 2020-09-21:
Do you mean an exception?
A breakpoint is not raised unless you set it

Huibean, 2020-09-22:
Hi @jonyMarino Thanks for you reply, sorry I am not familiar with visual studio, but I haven't change anything, it's like an exception but the visual studio it's a breakpoint

rajat2004, 2020-10-07:
You could perhaps try out #3064, the changes which it reverts might be causing the crash