microsoft/AirSim #3033: Add ability to switch views between agents in multi agent mode

nabeelsherazi, 2020-09-22:
Hi,

I was wondering if it would be possible to add a feature to Airsim allowing users to toggle between different agents (when there are multiple present), so that it is possible to supervise what each one is doing in swarm studies. Currently, each camera view, when switched, focuses only on the first agent created. There is no way to see what the other agents are doing unless they happen to come into the scene.

nabeelsherazi, 2020-10-03:
Hi @jonyMarino  thanks for adding the labels, but this is actually a feature request, not a question!

rajat2004, 2020-10-07:
Hmm, seems interesting, so you mean like switching the main viewport camera to follow different vehicles based on shortcut keys. Currently there's support for multi-vehicle in subwindows, it only allows viewing prespecified camera views though so limited in functionality

nabeelsherazi, 2020-10-08:
Correct! I'm currently trying to use Airsim to look at swarming behaviors, but it is a bit hard to know what other agents are doing. Maybe a shortcut key that would toggle the camera between each of the agents present in the sim (doing nothing if there's only 1 agent of course)?

Also, I think I could try tackling this myself, but I'm not at all familiar with the codebase so it would take me some time. If someone would know how to implement it from the get-go, that would definitely be better! Otherwise, if there are no takers, I will give it my best go :)

ViniciusAbrao, 2023-09-13:
Hi @nabeelsherazi ,

did you find a solution for your question?