microsoft/AirSim #3034: Adding AirSim to a blank project in order to control a drone. (VS2019, C++)

ItayKalfon, 2020-09-23:
I'm trying to control a drone using AirSim via Unreal Engine and visual studio. Before the real issue these are all of the specs:
OS - Windows 10
AirSim  - v1.3.1
Unreal Engine - v4.24.3
Visual Studio - 2019 Community Edition

I was able to install AirSim smoothly and the drone runs in the simulator and the sample project (Drone Shell) which is given in the AirSim repo is working fine.

I opened a completely new (C++) project in visual studio in order to try the code that was given in the C++ API tutorial. See code below. I also copied the AirLib folder into the solution directory so it would recognize it.
`#include <iostream>
#include "vehicles/multirotor/api/MultirotorRpcLibClient.hpp"

int main()
{
    using namespace std;
    msr::airlib::MultirotorRpcLibClient client;

    cout << "Press Enter to enable API control" << endl; cin.get();
    client.enableApiControl(true);

    cout << "Press Enter to arm the drone" << endl; cin.get();
    client.armDisarm(true);

    cout << "Press Enter to takeoff" << endl; cin.get();
    client.takeoffAsync(5)->waitOnLastTask();

    cout << "Press Enter to move 5 meters in x direction with 1 m/s velocity" << endl; cin.get();
    auto position = client.getMultirotorState().getPosition(); // from current location
    client.moveToPositionAsync(position.x() + 5, position.y(), position.z(), 1)->waitOnLastTask();

    cout << "Press Enter to land" << endl; cin.get();
    client.landAsync()->waitOnLastTask();

    return 0;
}`

I also added the following settings to the project properties:

- C/C++ -> General -> Additional Include Directories: `$(ProjectDir)..\AirLib\deps\rpclib\include;include;$(ProjectDir)..\AirLib\deps\eigen3;$(ProjectDir)..\AirLib\include`
- Linker -> General -> Additional Library Directories: `$(ProjectDir)..\AirLib\deps\MavLinkCom\lib\$(Platform)\$(Configuration);$(ProjectDir)..\AirLib\deps\rpclib\lib\$(Platform)\$(Configuration);$(ProjectDir)..\AirLib\lib\$(Platform)\$(Configuration)`
- Linker -> Input -> Additional Dependencies: `rpc.lib`

Visual Studio doesn't raise any errors in the code itself but when I compile it a lot of linking errors come up, all of them are LNK2019, and one example is given below.
`LNK2019	unresolved external symbol "public: bool __cdecl msr::airlib::RpcLibClientBase::armDisarm(bool,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?armDisarm@RpcLibClientBase@airlib@msr@@QEAA_N_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) referenced in function main	SimulatorTest	D:\Users\itay0\Desktop\Projects\Magshimim Project\SimulatorTest\SimulatorTest\Main.obj`

There is one error for each function used in the code (I can bring them too if necessary), I tried to include a lot of different directories but none helped.

Thanks ahead,
Itay.

jonyMarino, 2020-09-24:
Hi @ItayKalfon! I got the same errors. I solve it adding AirLib and MavLinkCom to the project references. Just right click `References` and then `Add reference...`, then select AirLib and MavLinkCom (wich must be in your solution).
In general I think is a better idea to copy and paste the HelloDrone project.
tell me how it went!

ItayKalfon, 2020-09-24:
Yeah it worked flawlessly. 
Thank you very much, the answer was spot on!

closed 2020-09-24 by ItayKalfon