microsoft/AirSim #3036: the fllowing modules are missing or built with a different engine version

ZhenWang-GoodST, 2020-09-23:
I got the “the fllowing modules are missing or built with a different engine version”， when I finished compile the airsim,  and the plugings directory becomes bigger, I download the mountain demo , and copy the plugins , modifies the uproject as the turtorial, how can I get the RIGHT version?

xxEoD2242, 2020-09-23:
What did you name your project?

ZhenWang-GoodST, 2020-09-23:
> What did you name your project?

I named the demo "mountain", and the airsim is the same with the document. I just follow the turtorial, vs is 2019 ,UE2.42, and I had tried the FAQS, but seems no use

ZhenWang-GoodST, 2020-09-24:
I ran the demo after several tries， maybe run the update_from_git.bat and the next command works, but I am confused why it didn't work in the beginning

closed 2020-09-24 by ZhenWang-GoodST