microsoft/AirSim #3037: Unable to stabilize only roll and pitch using gimbal

DanielGilo, 2020-09-23:
We face a similar issue to the one mentioned in #2140. We only want to stabilize pitch and roll (we don't want yaw stabilization) for the bottom_center camera (the one that looks down) of the drone. But when omitting the yaw from the settings.json (only specifying roll and pitch for the gimbal), the camera's yaw moves in an unstable way - doesn't follow the drone's orientation. 

[image]

If the settings are this way, we get stabilization on yaw (not what we want).

[image]

This is supposed to be what we what we want, but as we said the yaw in this case "goes crazy", we just want it to follow the drone's orientation.

Is there any solution?

Zartris, 2020-09-24:
I have the exact same problem! 
This is very far from ideal.

lachlanbrewster1, 2021-01-14:
Having the same issue, would be great to get this resolved. 

I want to stabilize roll and pitch while allowing yaw to rotate with the drone.

lachlanbrewster1, 2021-01-15:
I have found an okay for now work around if you intend to use the camera the same way as I do: Using the current image to move the drone relative to its own orientation. 

Record the drones initial yaw (at launch), then whenever you want to use the current camera frame for something, rotate it by the difference between your initial and current yaw, this is the true image relative to the drones orientation.

jprodriguezg, 2024-03-28:
I have the same issue reported above. 
It seems like the gimbal functionality only works while stabilizing all three rotation axes. I am wondering if someone is working on a solution at AirSim.

Zartris, 2024-03-28:
@jprodriguezg I can tell you for sure that is not the case. 
Airsim is a product at Microsoft now and is not being developed here anymore.
Some repo'es are trying to keep improving and keep the project alive, but even they have not seen updates for a while.
One of them is https://github.com/CodexLabsLLC/Colosseum which is the last one I have been using.

jprodriguezg, 2024-03-28:
I see. thanks for the feedback @Zartris !

Zartris, 2024-03-28:
@jprodriguezg No problem :)