microsoft/AirSim #3038: Error packaging AirSim for Android - fatal error: 'UdpSocket.hpp' file not found

dbaldwin, 2020-09-23:
I apologize if this is not the place for this, but there appears to be a bug when packaging AirSim for Android devices. Our goal is to get a build together with the Blocks environment and run some tests on Android OS and potentially Chrome OS. I realize Chrome OS may be a stretch, but we'll see.

We have no problems packaging AirSim blocks for Windows. The problem lies when trying to package for Android we ran into an issue related to 'UDPSocket.hpp' not being found. Here is the specific error:

```
UATHelper: Packaging (Android (ETC2)):     In file included from C:/Users/denni/Downloads/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Android/UE4/Development/AirSim/Module.AirSim.cpp:38:
UATHelper: Packaging (Android (ETC2)):     In file included from C:/Users/denni/Downloads/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/Car/CarPawnSimApi.cpp:1:
UATHelper: Packaging (Android (ETC2)):     In file included from C:/Users/denni/Downloads/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source\Vehicles/Car/CarPawnSimApi.h:14:
UATHelper: Packaging (Android (ETC2)):     In file included from C:/Users/denni/Downloads/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include\vehicles/car/CarApiFactory.hpp:8:
UATHelper: Packaging (Android (ETC2)):     C:/Users/denni/Downloads/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include\vehicles/car/firmwares/ardurover/ArduRoverApi.hpp(20,10): fatal error: 'UdpSocket.hpp' file not found
UATHelper: Packaging (Android (ETC2)):     #include "UdpSocket.hpp"
UATHelper: Packaging (Android (ETC2)):              ^~~~~~~~~~~~~~~
UATHelper: Packaging (Android (ETC2)):     2 warnings and 1 error generated.
PackagingResults: Warning: UObjectProperty has been renamed to FObjectProperty Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile. [-W#pragma-messages]
PackagingResults: Warning: UObjectProperty has been renamed to FObjectProperty Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile. [-W#pragma-messages]
PackagingResults: Error: 'UdpSocket.hpp' file not found
UATHelper: Packaging (Android (ETC2)):     [5/7] Module.AirSim.gen.cpp [armv7]
UATHelper: Packaging (Android (ETC2)): Took 936.758397s to run UnrealBuildTool.exe, ExitCode=6
UATHelper: Packaging (Android (ETC2)): UnrealBuildTool failed. See log for more details. (C:\Users\denni\AppData\Roaming\Unreal Engine\AutomationTool\Logs\C+Program+Files+Epic+Games+UE_4.25\UBT-Blocks-Android-Development_2.txt)
UATHelper: Packaging (Android (ETC2)): AutomationTool exiting with ExitCode=6 (6)
UATHelper: Packaging (Android (ETC2)): BUILD FAILED
```

We have successfully gone through the AirSim build instructions fro m the Wiki. We're running Windows 10, Visual Studio 2019, and Unreal 4.25. Spent hours on this trying to sort through this. The UdpSocket file comes from the MavLinkCom folder:

https://github.com/microsoft/AirSim/tree/master/MavLinkCom/include

But I don't know enough yet regarding how to include it in the build process. The fact that this works when packaging for Windows, but not Android leads me to believe there's a configuration issue somewhere. Has anyone here experienced this and would you be willing to shed some light? Attached is a screenshot of the build failure.

Many thanks in advance.

[image]

rajat2004, 2020-10-01:
Very, very strange. UdpSocket.hpp has been in AirSim since more than a year now, maybe no one has tried to do an Android build till now? Really no idea what could be causing this, while adding made sure that this was present in all the .vcxproj files etc, since the ArduPilot integration was initially done on Linux.
As a bad workaround, you could comment out the `ArduCopter` lines in [MultirotorParamsFactory](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParamsFactory.hpp), same for [CarApiFactory.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/car/CarApiFactory.hpp) and then maybe even delete the ArduPilot files altogether, they are the only ones using `UdpSocket.hpp`

dbaldwin, 2020-10-01:
Thanks @rajat2004 for the reply. I'll look into your suggestion. Let me add that I was able to package for Mac successfully and everything ran well, but when trying to package for iOS I ran into the exact same error. Maybe there is a different build configuration when targeting mobile devices. I'm new to Unreal Engine so I'm drinking from a firehose. I'll share what I find out.

[image]

rajat2004, 2020-10-02:
Might be good to post at Unreal Engine forums or something, people more familiar with UE4 and its quirks would have a better idea of whats going wrong

GaoGeolone, 2021-06-15:
@rajat2004 Excuse me, have you figure it out? I met the same problem days before.

emersonjr, 2021-06-22:
i'm also having the same problem, but trying to package for Linux

emersonjr, 2021-06-22:
i'm also having the same problem, but trying to package for Linux

@dbaldwin  did you succeed in packaging for Linux?

dbaldwin, 2021-06-22:
@emersonjr no I didn't but I haven't totally given up. I will revisit it soon, but definitely could not package for Android on my PC. I did not try a Linux build. A friend of mine has done it. I'll see if I can get him to chime in.

rhalaly, 2021-12-29:
I tried to compile custom env to Linux (from Windows) and got the same error. Is anyone solve it?

emersonjr, 2021-12-31:
i ended up packaging on Linux for Linux, it worked. Know this isn't a good solution for this issue but anyways does a work around.

rhalaly, 2022-01-01:
@emersonjr ho, that's said. Compiling Unreal Engine from source on Linux takes HOURS.