microsoft/AirSim #3043: Android package error when building ROS wrapper

yuyangch, 2020-09-27:
Ubuntu 16.04, UE4 Editor version 4.18, python 2.7.12

Trying to build the ros wrapper. Following the steps listed on https://microsoft.github.io/AirSim/airsim_ros_pkgs/   , until catkin build. then there is an error appearing relating to Android.

I do have Android SDK installed, maybe that played a part?

I tried to make the unreal engine again, but the error persists.

```
 ~/catkin_ws/src/AirSim/ros$ catkin build -v -DCMAKE_C_COMPILER=gcc-8 -DCMAKE_CXX_COMPILER=g++-8

 --------------------------------------------------------------------------------
 Profile:                     default
 Extending:          [cached] /opt/ros/kinetic
 Workspace:                   /home/chen/catkin_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/chen/catkin_ws/build
Devel Space:        [exists] /home/chen/catkin_ws/devel
Install Space:      [unused] /home/chen/catkin_ws/install
Log Space:          [exists] /home/chen/catkin_ws/logs
Source Space:       [exists] /home/chen/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_C_COMPILER=gcc-8 -DCMAKE_CXX_COMPILER=g++-8
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/bin/catkin", line 11, in <module>
    load_entry_point('catkin-tools==0.6.1', 'console_scripts', 'catkin')()
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 272, in main
    catkin_main(sysargs)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 267, in catkin_main
    sys.exit(args.main(args) or 0)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/cli.py", line 422, in main
    summarize_build=opts.summarize  # Can be True, False, or None
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 283, in build_isolated_workspace
    workspace_packages = find_packages(context.source_space_abs, exclude_subspaces=True, warnings=[])
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 89, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 150, in find_packages_allowing_duplicates
    xml, filename=filename, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 547, in parse_package_string
    raise InvalidPackage('The manifest must contain a single "package" root tag', filename)
catkin_pkg.package.InvalidPackage: Error(s) in package '/home/chen/catkin_ws/src/UnrealEngine/Engine/Source/ThirdParty/Android/package.xml':
The manifest must contain a single "package" root tag
```

GimpelZhang, 2020-09-27:
I suggest that Unreal Engine 4.24+ would be a better start. 

I mean using the lastest code (AirSim itself and the ROS wrapper) might not generate problems.

yuyangch, 2020-09-27:
I have since figured out the problem. It's my mistake to put both AirSim and UnrealEngine directories into my ROS workspace : catkin_ws/src/

worked around it by targeting the specific package
`catkin build airsim_ros_pkgs -DCMAKE_C_COMPILER=gcc-8 -DCMAKE_CXX_COMPILER=g++-8`
`catkin build airsim_tutorial_pkgs -DCMAKE_C_COMPILER=gcc-8 -DCMAKE_CXX_COMPILER=g++-8`

@GimpelZhang although not the problem here, but thanks for the advise nonetheless, i need to upgrade at some point

closed 2020-09-27 by yuyangch