microsoft/AirSim #3044: Propeller physics - simulation of trust/lift?

lkoelman, 2020-09-27:
I am trying to find out how the propellor rotation is linked to the drone's movement/kinematics.

It's not clear for me from the code (MultiRotor cpp classes) and BP_FlyingPawn blueprint how the propellor lift is simulated. Is this calculated entirely by the physics simulation purely based on the movement of the realistically shaped propellors? Or are the force vectors calculated in a simpler way somewhere else in the code?

rajat2004, 2020-10-01:
AFAIK, the Unreal Engine copter model isn't associated with the physics, the model is used for the rendering of the copter, props, etc but the physics is all handled inside FastPhysicsEngine only.
In Car vehicle, the physics is handled inside UE using PhysX

EdwinMeriaux, 2020-11-27:
@rajat2004 
So when designing our own custom drone is there anything to modify in the FastPhysicsEngine?

closed 2021-02-01 by jonyMarino