microsoft/AirSim #3045: How to output depth information to SITL?

BrellenZhou, 2020-09-27:
Hi, 

I want to output airsim's depth information to Ardupilot in SITL to achieve obstacle avoidance target. Ardupilot/PX4 has already have the computer-vision based obstacle avoidance algorithm. So I want to know how to output the depth to SITL to achieve this target. I cannot find the code part where the airsim output the depth/image info to the port.

I am very confusing. Could you please give some instructions? Thank you.

rajat2004, 2020-10-01:
You can use the [Image APIs](https://microsoft.github.io/AirSim/image_apis/) to fetch the depth images, and then generate obstacle information in a format which ArduPilot, PX4 can receive, such as [OBSTACLE_DISTANCE](https://mavlink.io/en/messages/common.html#OBSTACLE_DISTANCE), or whichever other messages or capabilities are supported

jonyMarino, 2020-11-06:
Hi @BrellenZhou! How it went?

closed 2021-04-30 by jonyMarino