microsoft/AirSim #3050: Error at startup: VehicleSetting for vehicle name BP_FlyingPawn_5 was requested but not found

PeterParkerLogic, 2020-09-29:
I followed the tutorial ( 9:27 in https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be) to create a drone by myself. 
But it turned out "Error at startup: VehicleSetting for vehicle name BP_FlyingPawn_5 was requested but not found".
I followed the method below, but it still not works.
https://www.gitmemory.com/issue/Microsoft/AirSim/1506/489733160

Please help me, Thanks

jonyMarino, 2020-09-29:
Hi! Do you get that error when tapping on play? My problem with that video is that I don't see that parameters on my Details menu

PeterParkerLogic, 2020-10-05:
Yes, I got this error, "VehicleSetting for vehicle name BP_FlyingPawn_5 was requested but not found", when I tapped on play.

jonyMarino, 2020-10-16:
I confirm this is happening on master (d59ceb7)

jonyMarino, 2020-11-20:
This PR #3146 fixed this issue!

closed 2020-11-20 by jonyMarino