microsoft/AirSim #3056: PX4 HIL simulation in the Block project  propellers don't  spining

on-chip-man, 2020-10-03:
win10 Home 1909
airsim 1.3.1
PX4 Releases 1.11.0 px4_fmu-v2_default
Unreal 4.25
python 3.8.5
VS comunity 2019

i already succeeded in block project fly the drone using simple-flight with python script ，and i also succeeded PX4 SITL simulation with WSL ubuntu 20.04, the drone just flying as the trajectory i give on the QGC.

however , when i change to HIL simulation(steps need to do wiith HIL simulation is less than SIL simulation), i can see throttle increased but the drone propeller didn't spin so the drone stay on the ground. When a hit the key ";" to see the reporter panel ，ue4 didn't Recieve any rotor input(rotor 1-4 input were all zoro). i tred several times result still the same.

i think there might be something wrong with the PX4 controller（PIX hawk HARDWARE）sending ACTUATOR message to ue4 or  ue4 project can't deal with the PWM out message,.I also check the mavlink message in QGC that PX4 PWM Output channel 1-4 are working well.

To conclusion，there are 2 main problems
1.Where can i find the source file name in PX4 firmware and airsim Block project that deal with sending and obtaining motor PWM signal, i plan to give some breakpoint or change the code to give siome debug report to find out the problem.there is another problem that the power parameter are lost and system power unavailable
[image]
[image]
[image]
[image]

2.VS community sometimes ran into exception such as in SimJoystick.cpp with null pointer exception and other exception reported as “ntdll.pdb not loaded”(Unhandled exception).generally I can click to continue run the project but don't know why that exception came out.

jonyMarino, 2021-06-01:
Please try our [new version](https://github.com/microsoft/AirSim/releases) which has many updates for px4.

lovettchris, 2021-06-04:
Also for HITL to work and send HIL_ACTUATOR_CONTROLS you need to use QGroundCOntrol to set the "Airframe" to HIL_QUADROCOPTER_X

lovettchris, 2021-07-08:
closing due to inactivity, please reopen if setting HIL_QUADROCOPTER_X did not solve your problem.

closed 2021-07-08 by lovettchris