microsoft/AirSim #3057: WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED

AIKUUM, 2020-10-03:
Details about my system:-
[image]
****************************************************************************************************************************************
I downloaded AirSim-master.zip from https://github.com/microsoft/airsim. I tried to execute the code in the file "hello_drone.py".
Contents of the code in "hello_drone.py":
import setup_path 
import airsim

import numpy as np
import os
import tempfile
import pprint
import cv2

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

state = client.getMultirotorState()
s = pprint.pformat(state)
print("state: %s" % s)

imu_data = client.getImuData()
s = pprint.pformat(imu_data)
print("imu_data: %s" % s)

barometer_data = client.getBarometerData()
s = pprint.pformat(barometer_data)
print("barometer_data: %s" % s)

magnetometer_data = client.getMagnetometerData()
s = pprint.pformat(magnetometer_data)
print("magnetometer_data: %s" % s)

gps_data = client.getGpsData()
s = pprint.pformat(gps_data)
print("gps_data: %s" % s)

airsim.wait_key('Press any key to takeoff')
client.takeoffAsync().join()

state = client.getMultirotorState()
print("state: %s" % pprint.pformat(state))

airsim.wait_key('Press any key to move vehicle to (-10, 10, -10) at 5 m/s')
client.moveToPositionAsync(-10, 10, -10, 5).join()

client.hoverAsync().join()

state = client.getMultirotorState()
print("state: %s" % pprint.pformat(state))

airsim.wait_key('Press any key to take images')
# get camera images from the car
responses = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
    airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, True), #depth in perspective projection
    airsim.ImageRequest("1", airsim.ImageType.Scene), #scene vision image in png format
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGBA array
print('Retrieved images: %d' % len(responses))

tmp_dir = os.path.join(tempfile.gettempdir(), "airsim_drone")
print ("Saving images to %s" % tmp_dir)
try:
    os.makedirs(tmp_dir)
except OSError:
    if not os.path.isdir(tmp_dir):
        raise

for idx, response in enumerate(responses):

    filename = os.path.join(tmp_dir, str(idx))

    if response.pixels_as_float:
        print("Type %d, size %d" % (response.image_type, len(response.image_data_float)))
        airsim.write_pfm(os.path.normpath(filename + '.pfm'), airsim.get_pfm_array(response))
    elif response.compress: #png format
        print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
        airsim.write_file(os.path.normpath(filename + '.png'), response.image_data_uint8)
    else: #uncompressed array
        print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
        img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) # get numpy array
        img_rgb = img1d.reshape(response.height, response.width, 3) # reshape array to 4 channel image array H X W X 3
        cv2.imwrite(os.path.normpath(filename + '.png'), img_rgb) # write to png

airsim.wait_key('Press any key to reset to original state')

client.armDisarm(False)
client.reset()

# that's enough fun for now. let's quit cleanly
client.enableApiControl(False)

***************************************************************************************************************************************
I'm using Unreal Engine 4.25.3
I executed the code in "hello_drone.py" using Spyder (Python 3.8). But I get the following error:
runfile('E:/AirSim/AirSim-master/AirSim-master/PythonClient/multirotor/hello_drone.py', wdir='E:/AirSim/AirSim-master/AirSim-master/PythonClient/multirotor')

WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED

WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED

WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED

WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
Traceback (most recent call last):

  File "E:\AirSim\AirSim-master\AirSim-master\PythonClient\multirotor\hello_drone.py", line 12, in <module>
    client.confirmConnection()

  File "E:\AirSim\AirSim-master\AirSim-master\PythonClient\airsim\client.py", line 130, in confirmConnection
    if self.ping():

  File "E:\AirSim\AirSim-master\AirSim-master\PythonClient\airsim\client.py", line 35, in ping
    return self.client.call('ping')

  File "D:\Anaconda\Anaconda_installation\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()

  File "D:\Anaconda\Anaconda_installation\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error

TransportError: Retry connection over the limit

WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED

rajat2004, 2020-10-03:
Did you actually start AirSim before running the script? If not, you can download one of the assets from the releases  -https://github.com/microsoft/AirSim/releases, or install Unreal Engine and build AirSim.
Also, side note, if posting code, please add fenced blocks like this - \``` code here \```` (triple backtick), makes it much easier to read. And no need to post the entire file if it's one the scripts already present in the repo, just linking to it is enough

AIKUUM, 2020-10-03:
In the link https://github.com/microsoft/AirSim/releases, I selected airsim v1.2.8. This led me to https://pypi.org/project/airsim/. I selected AirSim Package with Python APIs. This led me to https://github.com/microsoft/airsim. In https://github.com/microsoft/airsim, I selected code---download zip to download AirSim. Then I was facing the issue as I mentioned earlier.
************************************************************************************************************************************
I dint understand what you meant by "Did you actually start AirSim before running the script?". I might have done this. Can you please send a screenshot of what you had meant? Is it like executing some file in a specific directory of AirSim?

rajat2004, 2020-10-03:
That's just the Python API wrapper, AirSim is a simulator, so need to start that. Under `Assets`, there are zip files like `Blocks.zip`, etc. Download one of them, unzip and run the .exe in them. Then you can run the Python scripts
And see https://microsoft.github.io/AirSim/ for the documentation

AIKUUM, 2020-10-06:
I downloaded all the 11 windows assets from https://github.com/microsoft/AirSim/releases. I cant find python files in any of the zip files except in AirSim-1.3.1-windows.zip. In Blocks.zip, there is an exe file named Blocks.exe. When I run Blocks.exe, the simulator opens, and I can drive a car manually in an unrealistic environment. But when I try to execute "hello_drone.py" in "AirSim-1.3.1-windows.zip", I get the same error as I had mentioned earlier.

jonyMarino, 2020-10-06:
As rajat said there are 2 steps to run an airsim script:
1_ Start a simulation (for example run Blocks.exe (and if you want to use the drone tap on "no" in the prompted window))
2_ Run the script ( for example hello_drone.py if you are simulating a drone or  hello_car.py if you are simulating a car )

AIKUUM, 2020-10-07:
@jonyMarino, I'm aware of both the procedures which you have mentioned. But for running a script (for example:- hello_drone.py). I'm facing the issue as I had mentioned in https://github.com/microsoft/AirSim/issues/3057#issue-714024908

rajat2004, 2020-10-07:
If you have the simulation running and the example scripts still show the connection refused error, it might be a problem with the Windows Firewall or Defender, try disabling them and and test again.

wolf843, 2020-10-07:
I am currently running into the WSAECONNREFUSED error when using PythonClient as well (with disabling all firewalls). I can however execute all of the other scripts. Curious to if anyone has any solutions.

AIKUUM, 2020-10-08:
@wolf843, which other scripts are you talking about?? Python scripts?

wolf843, 2020-10-08:
@AIKUUM I was able to locate the cause of my issue... the port number passed into PythonClient did not match with the simulator port in my setup. Since I made the correction I have not experienced connection issues.

AIKUUM, 2020-10-09:
@wolf843, where do we check the simulator port? and In which file will the port number be mentioned in PythonClient folder?

AIKUUM, 2020-10-12:
> @AIKUUM I was able to locate the cause of my issue... the port number passed into PythonClient did not match with the simulator port in my setup. Since I made the correction I have not experienced connection issues.

@rajat2004, Can you mention the names of files in which such configurations should be done?

closed 2020-10-13 by AIKUUM

ProVival, 2021-08-29:
> @AIKUUM I was able to locate the cause of my issue... the port number passed into PythonClient did not match with the simulator port in my setup. Since I made the correction I have not experienced connection issues.

I am facing with the same issue but still couldn't find how to find the port that simulation uses and the port that PythonClient is trying to connect, not or how to change those ports. I would be very happy if you can help me