microsoft/AirSim #3059: server couldn't find function 'simSetCameraPose' with argument count 3

lucafei, 2020-10-05:
hello, i met a problem at calling function simSetCameraPose. the error can be seen in following:

[image]

ii have already checked the function simSetCameraPose is right and i built airsim in latest version. Do you know where is the problem? Thank you in advance.

ningzian, 2020-10-10:
I have got the same problem, and solved it by update airsim. You can try with https://microsoft.github.io/AirSim/unreal_custenv/#updating-your-environment-to-latest-version-of-airsim

jonyMarino, 2020-12-01:
@lucafei, can we close this?

lucafei, 2020-12-05:
yes, i closed this issue

closed 2020-12-05 by lucafei