microsoft/AirSim #3061: Running simulation synchronously with pwm and wind

p3jawors, 2020-10-06:
Ubuntu 16.04
Airsim latest master branch on commit 05a9625ca8391db50a152ad7882b08c02061f0b9
Python 3,5,2
Unreal 4.24.1-0+++UE4+Release-4.24

I have been running a simulation using pwm control, and recently added the new wind functionality. I have been using `client.moveByMotorPWMsAsync(pwm0, pwm1, pwm2, pwm3, dt).join()` to run the pwm control. I notice when I add a delay to my control loop (`time.sleep(0.01)`) my control starts to become unstable, which leads me to believe that I am not running synchronously. The function is specifically called Async, but it appears to have the `.join()` functionality. Is this the correct way to run this command in sync?

Secondly, is there an option for running `client.simSetWind()` synchronously? It does not appear to have the `.join()` functionality from what I can tell. It appears to be running asynchronously as well, but it is hard to tell if the instability I notice is from the pwm command, or if the wind command is also running asynchronously.

Any insights are greatly appreciated.

lucafei, 2020-10-08:
do you have already solved this problem now？ i have recently also met this.

p3jawors, 2020-10-09:
I'm noticing different flight performance under different computer loads. For example, if I have many chrome tabs open using ram and cpu, my quadrotor will not fly repeatedly. I will get different performance between the exact same tests. Under higher load it becomes completely unstable and can not fly at all. This is making me think that either, the simulation isn't deterministic, or things are running asynchronously somewhere. Maybe I'm missing when the simulation step is called, or if there are some seeds that I'm not setting, both of which could be causing the differences in performance.

jonyMarino, 2020-10-13:
May be this comment can be helpfull for you: https://github.com/microsoft/AirSim/issues/2974#issuecomment-683321580