microsoft/AirSim #3062: AirSim Blocks environment crashes on Ubuntu

Moj-Dev, 2020-10-07:
During running the [DQNdrone.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/DQNdrone.py) on Ubuntu18.04, UE4 crashes. I tried both versions 23 and 24, got the same error. Any ideas?

here is the crash report:
LoginId:62c709579e914f2a85c85fc9f38b709d-000003ec
EpicAccountId:

Assertion failed: Index < GetMaxIndex() [File:Runtime/Core/Public/Containers/SparseArray.h] [Line: 94]

libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
libUE4Editor-Core.so!AssertFailedImplV(char const*, char const*, int, char16_t const*, __va_list_tag*) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:100]
libUE4Editor-Core.so!FDebug::CheckVerifyFailedImpl(char const*, char const*, int, char16_t const*, ...) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:447]
libUE4Editor-Engine.so!TSparseArray<TSetElement<FTickFunction*>, TSparseArrayAllocator<TSizedDefaultAllocator<32>, FDefaultBitArrayAllocator> >::AllocateIndex(int) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Core/Public/Containers/SparseArray.h:94]
libUE4Editor-Engine.so!FSetElementId TSet<FTickFunction*, DefaultKeyFuncs<FTickFunction*, false>, FDefaultSetAllocator>::Emplace<FTickFunction* const&>(FTickFunction* const&&&, bool*) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Core/Public/Containers/Set.h:660]
libUE4Editor-Engine.so!FTickTaskLevel::AddTickFunction(FTickFunction*) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:1120]
libUE4Editor-Engine.so!FTickFunction::SetTickFunctionEnable(bool) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:1824]
libUE4Editor-Engine.so!UActorComponent::Activate(bool) [/home/mojtaba/UnrealEngine/Engine/Source/Runtime/Engine/Private/Components/ActorComponent.cpp:1697]
libUE4Editor-AirSim.so!APIPCamera::enableCaptureComponent(msr::airlib::ImageCaptureBase::ImageType, bool) [/home/mojtaba/AirSim/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/PIPCamera.cpp:460]
libUE4Editor-AirSim.so!UnrealImageCapture::getSceneCaptureImage(std::__1::vector<msr::airlib::ImageCaptureBase::ImageRequest, std::__1::allocator<msr::airlib::ImageCaptureBase::ImageRequest> > const&, std::__1::vector<msr::airlib::ImageCaptureBase::ImageResponse, std::__1::allocator<msr::airlib::ImageCaptureBase::ImageResponse> >&, bool) const [/home/mojtaba/AirSim/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/UnrealImageCapture.cpp:42]
libUE4Editor-AirSim.so!PawnSimApi::getImages(std::__1::vector<msr::airlib::ImageCaptureBase::ImageRequest, std::__1::allocator<msr::airlib::ImageCaptureBase::ImageRequest> > const&) const [/home/mojtaba/AirSim/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/PawnSimApi.cpp:195]
libUE4Editor-AirSim.so!msr::airlib::RpcLibServerBase::RpcLibServerBase(msr::airlib::ApiProvider*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short)::$_13::operator()(std::__1::vector<msr::airlib_rpclib::RpcLibAdapatorsBase::ImageRequest, std::__1::allocator<msr::airlib_rpclib::RpcLibAdapatorsBase::ImageRequest> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const [/home/mojtaba/AirSim/AirLib/src/api/RpcLibServerBase.cpp:139]
libUE4Editor-AirSim.so! [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/detail/call.h:75]
libUE4Editor-AirSim.so! [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/detail/call.h:56]
libUE4Editor-AirSim.so!rpc::detail::call_helper<2ul>::call<msr::airlib::RpcLibServerBase::RpcLibServerBase(msr::airlib::ApiProvider*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short)::$_13, std::__1::vector<msr::airlib_rpclib::RpcLibAdapatorsBase::ImageRequest, std::__1::allocator<msr::airlib_rpclib::RpcLibAdapatorsBase::ImageRequest> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >(decltype(call_helper<(2ul) - (1)>::call(fpfp0std::get<(2ul) - (1)>(fp0)std::forward<>(fp1))), msr::airlib::RpcLibServerBase::RpcLibServerBase(msr::airlib::ApiProvider*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short)::$_13, std::__1::tuple<std::__1::vector<msr::airlib_rpclib::RpcLibAdapatorsBase::ImageRequest, std::__1::allocator<msr::airlib_rpclib::RpcLibAdapatorsBase::ImageRequest> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >&) [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/detail/call.h:56]
libUE4Editor-AirSim.so! [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/detail/call.h:84]
libUE4Editor-AirSim.so!<name omitted> [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:74]
libUE4Editor-AirSim.so!std::__1::__invoke<(lambda at /home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:67:40) &, const clmdep_msgpack::v2::object &> [/usr/lib/llvm-8/bin/../include/c++/v1/type_traits:4353]
libUE4Editor-AirSim.so!std::__1::__invoke_void_return_wrapper<std::__1::unique_ptr<clmdep_msgpack::v1::object_handle, std::__1::default_delete<clmdep_msgpack::v1::object_handle> > >::__call<(lambda at /home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:67:40) &, const clmdep_msgpack::v2::object &> [/usr/lib/llvm-8/bin/../include/c++/v1/__functional_base:318]
libUE4Editor-AirSim.so!std::__1::__function::__alloc_func<(lambda at /home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:67:40), std::__1::allocator<(lambda at /home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:67:40)>, std::__1::unique_ptr<clmdep_msgpack::v1::object_handle, std::__1::default_delete<clmdep_msgpack::v1::object_handle> > (const clmdep_msgpack::v2::object &)>::operator() [/usr/lib/llvm-8/bin/../include/c++/v1/functional:1527]
libUE4Editor-AirSim.so!std::__1::__function::__func<(lambda at /home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:67:40), std::__1::allocator<(lambda at /home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/include/rpc/dispatcher.inl:67:40)>, std::__1::unique_ptr<clmdep_msgpack::v1::object_handle, std::__1::default_delete<clmdep_msgpack::v1::object_handle> > (const clmdep_msgpack::v2::object &)>::operator() [/usr/lib/llvm-8/bin/../include/c++/v1/functional:1651]
libUE4Editor-AirSim.so!std::__1::__function::__value_func<std::__1::unique_ptr<clmdep_msgpack::v1::object_handle, std::__1::default_delete<clmdep_msgpack::v1::object_handle> > (clmdep_msgpack::v2::object const&)>::operator()(clmdep_msgpack::v2::object const&) const [/usr/lib/llvm-8/bin/../include/c++/v1/functional:1799]
libUE4Editor-AirSim.so!std::__1::function<std::__1::unique_ptr<clmdep_msgpack::v1::object_handle, std::__1::default_delete<clmdep_msgpack::v1::object_handle> > (clmdep_msgpack::v2::object const&)>::operator()(clmdep_msgpack::v2::object const&) const [/usr/lib/llvm-8/bin/../include/c++/v1/functional:2347]
libUE4Editor-AirSim.so!rpc::detail::dispatcher::dispatch_call(clmdep_msgpack::v2::object const&, bool) [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/dispatcher.cc:46]
libUE4Editor-AirSim.so!rpc::detail::dispatcher::dispatch(clmdep_msgpack::v2::object const&, bool) [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/dispatcher.cc:22]
libUE4Editor-AirSim.so!rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::'lambda'()::operator()() const [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/detail/server_session.cc:70]
libUE4Editor-AirSim.so!void clmdep_asio::clmdep_asio_handler_invoke<rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::'lambda'()>(rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::'lambda'()&, ...) [/home/mojtaba/AirSim/external/rpclib/rpclib-2.2.1/dependencies/include/asio/handler_invoke_hook.hpp:68]
[... cut ...] struct, std::__1::default_delete<std::__1::__thread_struct> >, std::__1::function<void ()> >&, std::__1::__tuple_indices<>) [/usr/lib/llvm-8/bin/../include/c++/v1/thread:342]
libUE4Editor-AirSim.so!void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, std::__1::function<void ()> > >(void*) [/usr/lib/llvm-8/bin/../include/c++/v1/thread:352]
libpthread.so.0!UnknownFunction(0x76da)
libc.so.6!clone(+0x3e)

ysteen, 2020-10-07:
I was using airsim ros wrapper and got the same issue.
when I just open the airsim_node  with unreal engine this problem happened.
my environment is ubuntu 18.04 and UE4.25 with ros melodic

rajat2004, 2020-10-07:
Maybe try out #3064, that could be causing the crashes

ashraf-badr, 2020-12-03:
@ysteen @Moj-Dev 

I have the exact crash as you guys . Can you write how you solve it please ?

@rajat2004  
Sorry, I tried it but it does not work for me , I still get the same crash .

When I run a ROS node for Publishing the image and another one to Show it . After a few seconds or minutes unreal crashes .
I am expert so maybe i did not do all the changes that I have to do from  #3064 . so It will be appreciated if you guide me . Thanks

GimpelZhang, 2021-01-07:
Hi all, when I used the ros wrapper, I had the same error began with: 

Assertion failed: Index < GetMaxIndex() [File:Runtime/Core/Public/Containers/SparseArray.h]

(ubuntu 16.04, Unreal 4.24)

I did the following things:

* Under the Unreal folder, ran ./Setup.sh and ./GenerateProjectFiles.sh

* Under the AirSim folder, went back to a commit before Sep 3, and ran ./setup.sh and ./build.sh

Then the error didn't show up again.

zimmy87, 2021-01-07:
possibly related to #2012