microsoft/AirSim #3069: "moveByRC" doesn't work.

RinR0000, 2020-10-08:
Hello.

I'm having a problem, please help me.

I'm trying to control a drone with an xbox controller using the API, but the "moveByRC" in the API doesn't work.
The "moveByRC" in the API does not work.

I checked the API and found out that
It said not available by default.

How can I use it?

[image]

niekvleeuwen, 2020-10-08:
I have noticed while working with `moveByRC` that the drone does not respond when you have activated the API for the drone. Also, make sure that the fields `is_valid` and `is_initialized` in [RCData](https://github.com/microsoft/AirSim/blob/b7a65bb7f7a9471a2ec0ce6f573512b880d3197a/AirLib/include/common/CommonStructs.hpp#L259) are set to true. There is an example with manual mode for python [here](https://github.com/microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/PythonClient/multirotor/manual_mode_demo.py).

RinR0000, 2020-10-08:
@niekvleeuwen

Thank you for the quick answer.
I tried the code in the link you gave me, but
"moveByManualAsync" will raise it infinitely.

Also, the API code says it's not available by default, does that mean I can use ""moveByRC""?

niekvleeuwen, 2020-10-08:
Yes it is possible to use the `moveByRC` function in the current AirSim version. See also [this](https://github.com/microsoft/AirSim/blob/c340b7653efd22cefe2ae3f9110daf87d3fc6ae9/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightApi.hpp#L106) line where the code is implemented. Below is an example of working code (in c++).  
```cpp
#include "common/common_utils/StrictMode.hpp"
STRICT_MODE_OFF
#ifndef RPCLIB_MSGPACK
#define RPCLIB_MSGPACK clmdep_msgpack
#endif // !RPCLIB_MSGPACK
#include "rpc/rpc_error.h"
STRICT_MODE_ON

#include "vehicles/multirotor/api/MultirotorRpcLibClient.hpp"
#include <iostream>
#include <chrono>

int main() 
{
    using namespace msr::airlib;
    msr::airlib::MultirotorRpcLibClient client;
    RCData rc_data;
    
    try {
        client.confirmConnection();
        rc_data.is_initialized = true;
        rc_data.is_valid = true;

        rc_data.throttle = 1;
        client.moveByRC(rc_data);
        
        std::chrono::seconds timespan(3);
        std::this_thread::sleep_for(timespan);

        rc_data.throttle = 0.5;
        client.moveByRC(rc_data);
    }
    catch (rpc::rpc_error&  e) {
        std::string msg = e.get_error().as<std::string>();
        std::cout << "Exception raised by the API, something went wrong." << std::endl << msg << std::endl;
    }

    return 0;
}
```

RinR0000, 2020-10-10:
＠niekvleeuwen 

Sorry for the late reply.

I've moved the code you pushed, but the drone still won't work.

TakeoffAsync, etc. works, but
moveByRC does not work.

What could be the cause?

niekvleeuwen, 2020-10-15:
What does your `settings.json` look like? I have set the `RemoteControlID` to -1. 

```json
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "CameraDrone": {
            "VehicleType": "SimpleFlight",
            "X": 0,"Y": 0,"Z": 0,
            "RC": {
              "RemoteControlID": -1
            }
        }
    }
}
```

RinR0000, 2020-10-19:
＠niekvleeuwen

Sorry.

I used the setting.json and source code you told me about, but it didn't work.

yrik, 2023-11-20:
looks like I have the same issue, I've used proposed settings.json and code from https://github.com/microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/PythonClient/multirotor/manual_mode_demo.py but it does not work

yrik, 2023-11-21:
ok, the solution is to:
- do not enable api control mode
- use 
```
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "CameraDrone": {
            "VehicleType": "SimpleFlight",
            "X": 0,"Y": 0,"Z": 0,
            "RC": {
              "RemoteControlID": -1
            }
        }
    }
}
```
- make sure that you send correct controls (roll, pitch, yaw should be 0, throttle 1)

Mevaco2000, 2023-12-12:
> ok, the solution is to:
>
> * do not enable api control mode
> * use
>
> ```
> {
>     "SettingsVersion": 1.2,
>     "SimMode": "Multirotor",
>     "Vehicles": {
>         "CameraDrone": {
>             "VehicleType": [...]

Could you tell me why yaw should be 0? Isnt it possible to set yaw during using movebyRC command?