microsoft/AirSim #3071: Camera frame rate over ros topic changes if UE Editor is minimized.

m1baldwin, 2020-10-09:
I am testing the rate at which I can get images over rostopic from the airsim wrapper package. I noticed that when I run the simulation, run the airsim wrapper, and keep the Unreal Editor viewport open - on my screen - I get a reasonable FPS. 

However, if I then minimize the Unreal Editor window, the measured frame rate of the image topic begins to drop. If I click on the UE Editor to bring it back into the foreground, the measured update rate of the image topic begins to climb again. 

I notice a corresponding change in the GPU performance as well. I am guessing something is happening where having the UE Editor in the foreground is allowing full usage of the GPU, which in turn is allowing a higher update rate of the image data coming over ros topic. When UE Editor move to the background, GPU allocation drops significantly, and performance takes a hit. 

I am running this on a Windows 10 machine running WSL2 for the ROS part of things. Airsim/Unreal is running on the Windows host. I am measuring the update rate of the image topic using rostopic hz. I have UE configured to use the same amount of cpu regardless whether the application is in the fore/background.

I wonder if this is a windows thing trying to optimize the GPU usage?

yx0123, 2020-10-27:
Go to 'Edit->Editor Preferences' in Unreal Editor, in the 'Search' box type 'CPU' and ensure that the 'Use Less CPU when in Background' is unchecked. If you don't do this then UE will be slowed down dramatically when UE window loses focus. [Source](https://microsoft.github.io/AirSim/unreal_custenv/)

m1baldwin, 2020-10-27:
I will add a slight update to this; I have the "Use Less CPU when in Background" unchecked based on the instructions here: https://microsoft.github.io/AirSim/unreal_custenv/

So, this issue is happening even when Unreal is supposed to be using the same CPU whether it is in the foreground or background.