microsoft/AirSim #3072: closing of an issue

AIKUUM, 2020-10-10:
@jonyMarino, https://github.com/microsoft/AirSim/issues/3058#issue-714060232 issue has been closed. Please give freedom to the authors to close the issue. 
Well, I'm new to gitHub. So please let me know what the need was to close the issue.

AIKUUM, 2020-10-10:
@rajat2004, In regard to the issue, https://github.com/microsoft/AirSim/issues/3058#issue-714060232, I forgot to mention the other day. When I try to simulate a drone environment, I don't know how to fly a drone using keyboard. I can just change the view(ground observer, up, down, left, right etc.) with respect to the drone. Please let me know what keyboard controls I should use to fly a drone.

system1712, 2020-10-10:
#3058

jonyMarino, 2020-10-12:
Hi @AIKUUM! Rajat's answer seemed sufficient to me. If you have already read the documentation and still have questions, you can ask it in that issue and it will be reopened. I apologize if it was very drastic but in general these types of issues are forgotten in the repository for a couple of years.

AIKUUM, 2020-10-13:
> Hi @AIKUUM! Rajat's answer seemed sufficient to me. If you have already read the documentation and still have questions, you can ask it in that issue and it will be reopened. I apologize if it was [...]

I would never know when which person would come up with what types of answers. Its possible that I might get different types of answers from different people. So I dint close the issue. Referring to https://github.com/microsoft/AirSim/issues/3058#issue-714060232, as you can see @rajat2004 had told that its possible to fly a drone using keyboard controls. But according to https://github.com/microsoft/AirSim/issues/2698#issuecomment-630215275, its not possible to fly drones using keyboard controls as mentioned by @rajat2004. I don't know if some additional feature has been added for AirSim since then, such that keyboard controls can be used for flying drones.

rajat2004, 2020-10-13:
Yes, really sorry for the somewhat misleading response, keyboard control for drone is currently not possible. I was referring to using remote control for the drone control

closed 2020-10-13 by AIKUUM

jonyMarino, 2020-10-13:
@AIKUUM I apologize. I reopen the issue and close it when you see fit. We are here to help us!