microsoft/AirSim #3078: Press any key to takeoff

AIKUUM, 2020-10-13:
I opened the exe file in City.zip asset. Then I executed the hello_drone.py in PythonClient. Execution happens till printing of the statement "Press any key to takeoff". I tried pressing some keys. But nothing happened. Below the output of the execution:

_**_runfile('E:/AirSim/AirSim assets/AirSim-1.3.1-windows/AirSim-1.3.1-windows/PythonClient/multirotor/hello_drone.py', wdir='E:/AirSim/AirSim assets/AirSim-1.3.1-windows/AirSim-1.3.1-windows/PythonClient/multirotor')
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

state: <MultirotorState> {   'collision': <CollisionInfo> {   'has_collided': False,
    'impact_point': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'normal': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'object_id': -1,
    'object_name': '',
    'penetration_depth': 0.0,
    'position': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'time_stamp': 0},
    'gps_location': <GeoPoint> {   'altitude': 126.25707244873047,
    'latitude': 47.64558006836814,
    'longitude': -122.13335299735921},
    'kinematics_estimated': <KinematicsState> {   'angular_acceleration': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'angular_velocity': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'linear_acceleration': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'linear_velocity': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'orientation': <Quaternionr> {   'w_val': 0.9063065052032471,
    'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.422620952129364},
    'position': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 1.5934808254241943}},
    'landed_state': 0,
    'rc_data': <RCData> {   'is_initialized': False,
    'is_valid': False,
    'left_z': 0.0,
    'pitch': 0.0,
    'right_z': 0.0,
    'roll': 0.0,
    'switches': 0,
    'throttle': 0.0,
    'timestamp': 0,
    'vendor_id': '',
    'yaw': 0.0},
    'timestamp': 1602580569926508032}
imu_data: <ImuData> {   'angular_velocity': <Vector3r> {   'x_val': -2.4595410650363192e-05,
    'y_val': 0.0011013834737241268,
    'z_val': -0.0008111639181151986},
    'linear_acceleration': <Vector3r> {   'x_val': 0.06500411778688431,
    'y_val': -0.07115107774734497,
    'z_val': -9.836617469787598},
    'orientation': <Quaternionr> {   'w_val': 0.9063065052032471,
    'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.422620952129364},
    'time_stamp': 1602580569938508288}
barometer_data: <BarometerData> {   'altitude': 126.20809173583984,
    'pressure': 99817.390625,
    'qnh': 1013.25,
    'time_stamp': 1602580569938508288}
magnetometer_data: <MagnetometerData> {   'magnetic_field_body': <Vector3r> {   'x_val': 0.1907837688922882,
    'y_val': -0.17729032039642334,
    'z_val': 0.3700033128261566},
    'magnetic_field_covariance': [   ],
    'time_stamp': 1602580569938508288}
gps_data: <GpsData> {   'gnss': <GnssReport> {   'eph': 0.3000044822692871,
    'epv': 0.40000447630882263,
    'fix_type': 3,
    'geo_point': <GeoPoint> {   'altitude': 126.25707244873047,
    'latitude': 47.64558006836814,
    'longitude': -122.13335299735921},
    'time_utc': 1602580569722503,
    'velocity': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0}},
    'is_valid': True,
    'time_stamp': 1602580569722503680}
Press any key to takeoff_**_

Output in unreal engine:
[image]

I would appreciate it if some one could help me with how to proceed further.

AIKUUM, 2020-10-20:
> I opened the exe file in City.zip asset. Then I executed the hello_drone.py in PythonClient. Execution happens till printing of the statement "Press any key to takeoff". I tried pressing some keys. But nothing happened. Below the output of the execution:
> 
> _**_runfile('E:/AirSim/AirSim assets/AirSim-1.3.1-windows/AirSim-1.3.1-windows/PythonClient/multirotor/hello_drone.py', wdir='E:/AirSim/AirSim assets/AirSim-1.3.1-windows/AirSim-1.3.1-windows/PythonClient/multirotor')
> Connected!
> Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)
> 
> state: { 'collision': { 'has_collided': False,
> 'impact_point': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'normal': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'object_id': -1,
> 'object_name': '',
> 'penetration_depth': 0.0,
> 'position': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'time_stamp': 0},
> 'gps_location': { 'altitude': 126.25707244873047,
> 'latitude': 47.64558006836814,
> 'longitude': -122.13335299735921},
> 'kinematics_estimated': { 'angular_acceleration': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'angular_velocity': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'linear_acceleration': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'linear_velocity': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0},
> 'orientation': { 'w_val': 0.9063065052032471,
> 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.422620952129364},
> 'position': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 1.5934808254241943}},
> 'landed_state': 0,
> 'rc_data': { 'is_initialized': False,
> 'is_valid': False,
> 'left_z': 0.0,
> 'pitch': 0.0,
> 'right_z': 0.0,
> 'roll': 0.0,
> 'switches': 0,
> 'throttle': 0.0,
> 'timestamp': 0,
> 'vendor_id': '',
> 'yaw': 0.0},
> 'timestamp': 1602580569926508032}
> imu_data: { 'angular_velocity': { 'x_val': -2.4595410650363192e-05,
> 'y_val': 0.0011013834737241268,
> 'z_val': -0.0008111639181151986},
> 'linear_acceleration': { 'x_val': 0.06500411778688431,
> 'y_val': -0.07115107774734497,
> 'z_val': -9.836617469787598},
> 'orientation': { 'w_val': 0.9063065052032471,
> 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.422620952129364},
> 'time_stamp': 1602580569938508288}
> barometer_data: { 'altitude': 126.20809173583984,
> 'pressure': 99817.390625,
> 'qnh': 1013.25,
> 'time_stamp': 1602580569938508288}
> magnetometer_data: { 'magnetic_field_body': { 'x_val': 0.1907837688922882,
> 'y_val': -0.17729032039642334,
> 'z_val': 0.3700033128261566},
> 'magnetic_field_covariance': [ ],
> 'time_stamp': 1602580569938508288}
> gps_data: { 'gnss': { 'eph': 0.3000044822692871,
> 'epv': 0.40000447630882263,
> 'fix_type': 3,
> 'geo_point': { 'altitude': 126.25707244873047,
> 'latitude': 47.64558006836814,
> 'longitude': -122.13335299735921},
> 'time_utc': 1602580569722503,
> 'velocity': { 'x_val': 0.0,
> 'y_val': 0.0,
> 'z_val': 0.0}},
> 'is_valid': True,
> 'time_stamp': 1602580569722503680}
> Press any key to takeoff_**_
> 
> Output in unreal engine:
> [image]
> 
> I would appreciate it if some one could help me with how to proceed further.

@rajat2004, Can you give me a hint to proceed?

rajat2004, 2020-10-20:
@AIKUUM It's working fine for me on Linux with the latest binary, I'm not on Windows currently so can't test that though.
The data is appearing, so the API is working. Only thing I can think of is that the settings might have been modified, such as `AllowAPIAlways` being set to `false` in settings.json. Or the script being modified and the vehicle is not armed.

AIKUUM, 2020-10-27:
@rajat2004, Which key did you press for takeoff?

rajat2004, 2020-10-27:
Any key works as mentioned

AIKUUM, 2020-10-28:
@rajat2004, below is the code for "hello_drone.py":
``
import setup_path 
import airsim

import numpy as np
import os
import tempfile
import pprint
import cv2

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

state = client.getMultirotorState()
s = pprint.pformat(state)
print("state: %s" % s)

imu_data = client.getImuData()
s = pprint.pformat(imu_data)
print("imu_data: %s" % s)

barometer_data = client.getBarometerData()
s = pprint.pformat(barometer_data)
print("barometer_data: %s" % s)

magnetometer_data = client.getMagnetometerData()
s = pprint.pformat(magnetometer_data)
print("magnetometer_data: %s" % s)

gps_data = client.getGpsData()
s = pprint.pformat(gps_data)
print("gps_data: %s" % s)

airsim.wait_key('Press any key to takeoff')
client.takeoffAsync().join()

state = client.getMultirotorState()
print("state: %s" % pprint.pformat(state))

airsim.wait_key('Press any key to move vehicle to (-10, 10, -10) at 5 m/s')
client.moveToPositionAsync(-10, 10, -10, 5).join()

client.hoverAsync().join()

state = client.getMultirotorState()
print("state: %s" % pprint.pformat(state))

airsim.wait_key('Press any key to take images')
# get camera images from the car
responses = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
    airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, True), #depth in perspective projection
    airsim.ImageRequest("1", airsim.ImageType.Scene), #scene vision image in png format
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGBA array
print('Retrieved images: %d' % len(responses))

tmp_dir = os.path.join(tempfile.gettempdir(), "airsim_drone")
print ("Saving images to %s" % tmp_dir)
try:
    os.makedirs(tmp_dir)
except OSError:
    if not os.path.isdir(tmp_dir):
        raise

for idx, response in enumerate(responses):

    filename = os.path.join(tmp_dir, str(idx))

    if response.pixels_as_float:
        print("Type %d, size %d" % (response.image_type, len(response.image_data_float)))
        airsim.write_pfm(os.path.normpath(filename + '.pfm'), airsim.get_pfm_array(response))
    elif response.compress: #png format
        print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
        airsim.write_file(os.path.normpath(filename + '.png'), response.image_data_uint8)
    else: #uncompressed array
        print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
        img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) # get numpy array
        img_rgb = img1d.reshape(response.height, response.width, 3) # reshape array to 4 channel image array H X W X 3
        cv2.imwrite(os.path.normpath(filename + '.png'), img_rgb) # write to png

airsim.wait_key('Press any key to reset to original state')

client.armDisarm(False)
client.reset()

# that's enough fun for now. let's quit cleanly
client.enableApiControl(False)
``
Is the code alright? Or is there some mistake in the code because of which the takeoff is not happening even when some key is pressed?

closed 2020-10-30 by AIKUUM