microsoft/AirSim #3079: Implementing custom Airsim environment from your release especially on Linux

Melvin-leo, 2020-10-14:
Hi, one major question here,

you mentioned about custom environments like mountain landscape or soccer field that are included in the release v1.3.1. How are we going to create this custom environment. Even if we follow https://microsoft.github.io/AirSim/unreal_custenv/, the environments available are only in the marketplace, to create project. How are we going to use the custom environment airsim provides (that are not found in Epic launchers's marketplace)? How are we going to use the binaries files in Linux to create these custom environment.

jonyMarino, 2020-11-16:
Hi @Melvin-leo! I want to understand your necessity. Why are the binaries not suitable for you to use?

Melvin-leo, 2020-11-16:
It's not that it's not suitable , but there is no tutorial or instructions on how to implement the Linux binaries.

jonyMarino, 2020-11-17:
It is an all in one package. You need to run the .sh script, and you would have the environment and the plugin already running. (v1.4 is another story, but it is a draft so far)

closed 2021-01-04 by jonyMarino