microsoft/AirSim #3080: Ardupilot with Hardware in the Loop

blakermchale, 2020-10-14:
I am currently running unreal engine and using ArduPilot on a Pixhawk 4. I saw there was support for setting up hardware in the loop (HITL) with PX4, but could not find anything on ArduPilot support besides the software in the loop (SITL). Has anyone configured Ardupilot for HITL with AirSIm? If so, how did you do that and what control ports did you set in the settings.json?

bravepaulo, 2020-10-16:
Hi @blakermchale.
Have you tried to use the same parameters that are proposed in PX4 example?
I have worked with HITL using Airsim and PX4, maybe the settings on json file work out for ADP as well.

blakermchale, 2020-10-17:
Hi @bravepaulo,
I tried that but it returns back `Error while receiving rotor control data - ErrorNo: -1`. One step that I cannot find an alternative for is the `selecting the HIL Quadrocopter X airframe` for pixhawk 4. Any ideas?

rajat2004, 2021-01-18:
ArduPilot hasn't supported HITL since many years

closed 2021-04-22 by jonyMarino