microsoft/AirSim #3084: I want to

kensuke1921, 2020-10-18:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

weigangwen, 2020-10-19:
I think the record button may help you to solve this problem.When the car is running, click the red button at the bottom right corner of the screen.

jonyMarino, 2020-10-19:
with getCarState() you get a state instance with a speed attribute. You can  then save to a file or list