microsoft/AirSim #3087: Unity PX4 SITL Accel and Gyro Timeout

lberridge1992, 2020-10-19:
I've been trying to get PX4 SITL working with Airsim using Unity and have encountered an issue with gyro and accelerometer timeouts happening as soon as the sim is connected (see attached image). I'm a bit stumped by this so thought I would ask here. I'm using the latest Airsim master branch and PX4 firmware v1.10.1, both are running on a Windows VM, cygwin toolchain for executing px4 sitl. 

From what I gathered this should work, unless I'm missing something as most of the information is all related to Unreal and I'm unsure if SITL is supported for Unity.

[image]

I get the followings messages reported by Airsim in the Unity log file (apologies for the vehicle name being SimpleFlight, I've put this in place to work around an issue in the settings read in Unity which has SimpleFlight hardcoded as the name)

Connected to SITL over TCP. 30 Vehicle=SimpleFlight
Connecting to PX4 Control UDP port 14580, local IP 127.0.0.1, remote IP... 30 Vehicle=SimpleFlight
Ground control connected over UDP. 30 Vehicle=SimpleFlight
Enabling lockstep mode 30 Vehicle=SimpleFlight
Got GPS lock 30 Vehicle=SimpleFlight
Accel #0 fail:  TIMEOUT! 30 Vehicle=SimpleFlight
Gyro #0 fail:  TIMEOUT! 30 Vehicle=SimpleFlight

Settings are:
{
  "SimMode": "Multirotor",
  "SettingsVersion": 1.2,
  "ApiServerPort": 5000,
  "Vehicles": {
    "SimpleFlight":{
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 51.450711,
                "LPE_LON": -2.603829,
                "COM_OBL_ACT": 1
	}
    }
  }
}

jonyMarino, 2020-10-19:
Hi! May be this is not related with Unity. We have an incompatibility with that PX4 version for the moment https://github.com/microsoft/AirSim/issues/2477. May be this is a dup?

lberridge1992, 2020-10-20:
Hi Jony, thanks for the reply and heads up. I've actually tried running the same PX4 config with the built-in simulator jmavsim and get the same timeout messages on my setup too so expect it's something to do with my VM config rather than Airsim. Cranking the time-out values up stops the errors being displayed but then I get unstable flight assumingly because the sensors aren't updating quickly enough. Should Airsim be running with PX4 in lockstep mode? Most of the solutions to this problem on the PX4 forums seem to suggest turning lockstep off.

lberridge1992, 2020-11-23:
Had a dig around and answered my question: Airsim supports both lockstep and normal simulator modes, closing issue.

closed 2020-11-23 by lberridge1992