microsoft/AirSim #3089: How to 'kill' a drone programatically

mhl787156, 2020-10-20:
Hi again! 

I'm working on building a reliability and redundancy test platform for multi-uav systems. As part of this I need to simulate failures. Whilst I could get complicated, currently I just want something to get a particular drone to drop out the sky. 

Is there a particular function I could use which could interrupt everything and get the drone to fall? 

Is this something I'd have to modify AirLib, or would I be able to hack together something from the python client lib/ros

I checked the old issues, and I dont think this has come up before, except possibly in the context of either physical collisions or incorporating wind.

Many Thanks!

rajat2004, 2020-10-21:
Hmm, interesting
Drone falling out of the sky could essentially be all the motors stopping, could try using [movebymotorPWMs](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L908) and set everything to 0

mhl787156, 2020-10-21:
Okay yeah, I'll give that a go,

Just to check, I'm currently using airsim_ros_wrapper + ros however I want some of the functionality offered by the python client. It's a bit of a hack, but I have basically wrapped the python client in a rosnode exposing a few services for things like get_object_pose. 

Is this safe to use at the same time as the airsim_ros_wrapper? I dont know how the server treats multiple connections is all! So far it seems to work okay.