microsoft/AirSim #3095: AirSim with ROS WSL2 Connection Error

blakermchale, 2020-10-22:
I've been trying to connect my WSL2 ROS instance to AirSim on Windows. When I run the command `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP` it gets connected, but starts outputting this error:

```
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'SimpleFlight' is not available. This could either because this is not a simulation or this vehicle does not exist.
```

Currently, I am using Ubuntu 20.04 and have confirmed that the ros packages work on another linux computer running only Ubuntu 20.04.

Has anyone ran into this issue and resolved it?

t-thanh, 2020-10-29:
I have same issue - with ROS melodic in WSL Ubuntu 18.04 in Windows 10

In WSL, I set
ROS_IP = '172.27.154.160'
ROS_MASTER_URI = 'http://172.27.154.160:11311'

in Windows 10: IP: 172.27.144.1

roslauch is similar to @blakermchale : `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=172.27.154.160`

xistix, 2020-11-11:
Is there any progress on this? I have the same issue.

xistix, 2020-11-11:
I found this has something to do with the vehicle name in AirSim settings.json.
If you change the name like this:
...
"Vehicles": {
    "ComputerVision": {
...

the error goes away.

t-thanh, 2020-11-12:
@xistix Can you send me your settings.json? I tried to change it like you describe but it doesn't work as expected.

blakermchale, 2020-11-12:
@xistix Thanks! I was able to get it working by changing the name to "SimpleFlight". You also have to make sure that there is a settings.json in your WSL Documents/AirSim directory

jonyMarino, 2020-12-11:
Hi all! Thank you for your discussion here. I have submitted a document update through you #3194!is anyone interested in submitting a pull request to solve the drone name issue? Or maybe dig more into it?

closed 2020-12-15 by jonyMarino