microsoft/AirSim #3097: how to use the sepcified setting.json?

lucafei, 2020-10-23:
hello, i met a problem at using airsim. By now i installed two different versions of airsim. One is installed by using pip install airsim=1.3. Another version is generated by rebuilding the airsim, because i want to use the latest feature. But i find when i run unreal environment with different version airsim plugin. it always use the different setting.json file. for the with pip installed airsim it uses the setting.json which in /documents on windows. but for the rebuilded airsim it doesn't use this same setting.json. But i don't find where the setting.json which the rebuilded airsim uses is. For me i want to use a same setting.json. Do you guys know how to solve this problem. Thank you very much in advance.

ykhedar, 2020-10-30:
I think this is due to the pull request #3053.

jonyMarino, 2020-11-06:
Hi @lucafei! Does the used settings.json not appear on your log messages at the left of your simulation screen?